raspiblitz/raspiblitz #700: RTL hangs on RPI idle

woeisme, 2019-08-02:
I've noticed several times now that when the Raspberry PI (4) is left running for a few hours that RTL webhost becomes unresponsive. It's only after moving the menu items around via ssh raspiblitz menu that RTL becomes accessible again and I can log in to it again. This is all via local wired home network.

openoms, 2019-08-03:
Are you saying saying that RTL runs fine after a restart and only after  a while becomes unresponsive?

Check what
 `sudo systemctl status RTL` 
shows then.
You  can restart just RTL with:
`sudo systemctl restart RTL`
and check if that helps.

I opened and issue with them because had a problem installing RTL on the latest build using an Odroid HC1. See here: https://github.com/ShahanaFarooqui/RTL/issues/173

woeisme, 2019-08-03:
it becomes unresponsive in the browser, merely moving the cursor around in the ssh menu (raspiblitz main menu) "wakes it up" and the web site becomes reachable again.

openoms, 2019-08-06:
> I've noticed several times now that when the Raspberry PI (4) is left running for a few hours that RTL webhost becomes unresponsive. It's only after moving the menu items around via ssh raspiblitz [...]

Now I have made RTL work on the 1.3RC2 I can confirm this issue on my Odroid HC1. RTL website ws not respondig until I logged in via ssh (did not even need to get to the menu). Then works again. Running the latest RTL 0.4.4.

nyxnor, 2021-04-18:
Please close this.

closed 2021-04-19 by rootzoll