raspiblitz/raspiblitz #702: Use more robust way of detecting image type

Kixunil, 2019-08-02:
In two instances (dietpi, tegra) the image is detected by checking hostname. This is fragile as it breaks when the user sets his own hostname. It'd be better to detect it using some other way.

rootzoll, 2019-08-05:
@openoms can you maybe take a look?

openoms, 2019-08-05:
> In two instances (dietpi, tegra) the image is detected by checking hostname. This is fragile as it breaks when the user sets his own hostname. It'd be better to detect it using some other way.

Relevant code from build_sdcard.sh: 
```
echo "Detect Base Image ..."
baseImage="?"
isDietPi=$(uname -n | grep -c 'DietPi')
isRaspbian=$(cat /etc/os-release 2>/dev/null | grep -c 'Raspbian')
isArmbian=$(cat /etc/os-release 2>/dev/null | grep -c 'Debian')
isUbuntu=$(cat /etc/os-release 2>/dev/null | grep -c 'Ubuntu')
isNvidia=$(uname -a | grep -c 'tegra')
```

To detect the Nvidia Jetson Nano image we rely on the CPU chipset name being unique, see: https://github.com/openoms/raspiblitz/pull/50#issuecomment-497263480. This allows us to make a specific change during build only relevant to the factory supplied image to that SBC: https://github.com/openoms/raspiblitz/pull/50 . I don`t think we need to change that until a test on a Jetson Nano fails. It does not affect other builds and difficult to cheat.

Regarding the DietPi builds I am not planning to test more, more favour of Armbian now. As I have not seen anyone using it we could remove it, but can`t see any harm leaving it in either as an experimental option. 
Detecting by the hostname would only be be an issue if someone changes the hostname after burning a new SD with a custom image before running the build_sdcard.sh script. Changing the hostname afterwards does not trigger this script. Suggestions to detect the DietPi image in a more robust way are welcome.

nyxnor, 2021-04-18:
Pls close this, Openoms explained above.

Kixunil, 2021-04-18:
Ah, I lost notification for this one.

> we rely on the CPU chipset name being unique,

That makes sense, the linked issue suggests that `-tegra` is added to kernel release, so maybe use `-r` instead of `-a` to be more specific. (Not that I expect anyone to name their device `tegra`, but it's one char change to eliminate the risk.)

>  leaving it in either as an experimental option.

Should be documented that it's experimental

> if someone changes the hostname after burning a new SD 

Whatever can fail will fail, I've seen various weird things. :)

nyxnor, 2021-04-18:
@Kixunil Thanks for responding. 

>That makes sense, the linked issue suggests that -tegra is added to kernel release, so maybe use -r instead of -a to be more specific. (Not that I expect anyone to name their device tegra, but it's one char change to eliminate the risk.)

If it narrows down the possibility of error, should be done also.

>Should be documented that it's experimental

Yes, it is lacking information about being experimental. As much as I would like to edit FAQ and README, Im waiting for the wiki to be able to organize more.

closed 2021-11-08 by openoms