raspiblitz/raspiblitz #704: Proplems with initial setup on 1.3 rc1

CryptoSigmund, 2019-08-04:
I'm coming a long road of failures and help from openoms now here:

https://termbin.com/u0s1

Hardware:

PowerSupply:
Original Raspberry 4 power supply is to weak - it never passed the test so I sorted it out and only used the Apple USB-C - Powersupply - 29 Watt (MJ262Z/A) - cable = Apple usb-c cable that came with the powersupply.

Hardware is a Rasperry Pi 4 with 4GB Ram (produced in June)
USB 3.1 Conector Sabrent (EC-SS31)
SSD Samsung 1TB Pro 950

Hardware is a Rasperry Pi 4 with 4GB Ram (produced in Mai)
USB 3.0 Conector Sabrent (EC-SSHD)
SSD SanDisk SSD PLUS 1TB

Also try now a HDD Conector
USB 3.0 Inateck UA 1001 with external power supply

All ssd had been surfaced checked by Minitool Partition Wizard
SD Cards are several Samsung microSDHC pro Endurance 32gb

As the time I'm writing this... I have one bigger power supply from apple which I will switch just to be sure.

I build the software with:
-1.3 rc1 iso from rootzoll github
-1.3 from openoms github
all from scratch with the original raspian buster image

Flashing is done with balenaEtcher 1.5.52

Sync through Tor
IBD only from other nodes

In the blockchain sync I also get a flashing failure like another one from the Raspibolt telegram channel gets:
[image]

So I'm out of ideas what to do - it just get stucked in every combination of the setup process with a "bitcoind is not running" at some time.

CryptoSigmund, 2019-08-04:
one more issue pop up
[image]

fluidvoice, 2019-08-04:
The read-only file system error is probably caused by your previously insufficient power supply.  You may have to reformat your external hard drive and install/setup from scratch.

CryptoSigmund, 2019-08-04:
Sadly not. The last one was a fresh build with the biggest usb c power supply from apple.

fluidvoice, 2019-08-04:
> Sadly not. The last one was a fresh build with the biggest usb c power supply from apple.

You're running Tor. And I see lots of warnings in your log like this:

> Aug  4 08:29:25 CryptoSigmund tor[676]: Aug 04 08:29:25.000 [warn] /run/tor is not owned by this user (bitcoin, 1002) but by debian-tor (111). Perhaps you are running Tor as the wrong user?
> Aug  4 08:29:25 CryptoSigmund tor[676]: Aug 04 08:29:25.000 [warn] Before Tor can create a control socket in "/run/tor/control", the directory "/run/tor" needs to exist, and to be accessible only by the user and group account that is running Tor.  (On some Unix systems, anybody who can list a socket can connect to it, so Tor is being careful.)

This warning/problem was reported here a while ago. Perhaps and old problem that cropped up again: https://github.com/rootzoll/raspiblitz/issues/402#issue-417282446

fluidvoice, 2019-08-04:
It's also an open issue here: https://github.com/rootzoll/raspiblitz/issues/592#issuecomment-516017690

CryptoSigmund, 2019-08-04:
Never tried it without tor couse this is not an option. But it is reasonable.

Could this also be related to the rpi 4 usb Problem https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931 ?

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll