raspiblitz/raspiblitz #706: Rely only on GPG keys for verification

Kixunil, 2019-08-05:
GPG verification should be sufficient and the hashes are redundant and prevent smooth upgrades (the script must be changed). They only provide protection against the signer of the software being compromised.

If the protection against signer compromise is desired, multisig might be implemented instead.

rootzoll, 2019-08-05:
The latest PR from @openoms is using for the hashes the manifest file now - see:
https://github.com/rootzoll/raspiblitz/pull/696/files

As the LND documentation advises the manifest file with the hashes is the only one signed and gets checked against the GPG keys: https://github.com/LightningNetwork/lnd/releases

For the bitcoin-core we could get rid of the extra static hash values also in the build_script - thats something to consider.

Kixunil, 2019-08-10:
Yes, Bitcoin core is the one I had in mind.

rootzoll, 2019-08-11:
With the update to bitcoin core 0.18.1 I changed that the hash values will get greped from the sign checked asc file. Will be part of v1.3 release.

closed 2019-09-03 by rootzoll