raspiblitz/raspiblitz #707: Support of USP: APC over USB

rootzoll, 2019-08-05:
A UPS (Uninterruptible Power Supply) is used to protect the RaspiBlitz against power outages. Normally you put it just between your normal power outlet and your RaspiBlitz and you are good. But some UPS offer a way to communicate with devices. This can be very useful for example if on a longer power outage the battery of the UPS runs low the RaspiBlitz could detect this and power down in a clean way - instead of just the power goes out and risking data loss or corruption.

There is an experimental script to connect the RaspiBlitz to a UPS over USB cable build by APC - the Model tested with was APC Back-UPS BX - BX700U-GR --> https://www.amazon.de/APC-Back-UPS-Unterbrechungsfreie-Stromversorgung-BX700U-GR/dp/B00T7BYRCK but it should work with every APC model offering a USB port.

To turn it on run from terminal: `/home/admin/config.scripts/blitz.ups.sh on apcusb`

If you have other UPS models or ways to connect ... feel free to extend this script.

rootzoll, 2019-08-05:
TODO: Test APCUPSD running shutdown script when power goes out

openoms, 2019-08-05:
Very interesting!  I have this APC UPS model: https://www.amazon.co.uk/dp/B002RXED6A, but no cable yet. It look s like it would take something like this (RJ45 - USB cable): https://www.ebay.com/itm/ORIGINAL-AP9827-SIMPLE-SIGNALLING-USB-CABLE-FOR-APC-BACK-UPS-ES-G-BE450G-ES550/332806306495, will need to try.

The RaspiBlitz is getting to be the most resilient little node out there!

rootzoll, 2019-08-06:
@openoms The script has at the moment just the "apcusb" mode in witch the "apcupsd" service gets configured to listen on the usb-cable. Not sure how the LAN-2-USB cable is working you linked. If it needs another configuration - feel free to extend the script with a "apclanusb" mode or something :)

rootzoll, 2019-08-06:
Looks OK so far for the v1.3 release as a experimental commandline configuration. Already opened issue #710 for the improvements for future releases.

openoms, 2019-08-06:
@rootzoll this Rj45 to USB cable does not look to have ethernet function. It is just a simple signal cable, hopefully similar to yours. Will need to try.

Kixunil, 2019-08-10:
When I was studying UPS documentation a while ago, I learned that when low battery event comes, the system must be shutdown very quickly. Bitcoind may take a long time to stop (I noticed fixed 10s delay in the shutdown script - I don't think this is a good idea).

I think that it'd be better to start shutting down the machine sooner, but not sure how to reliably decide when. I'd be also useful to issue some kind of notification when power goes down.

rootzoll, 2019-08-12:
At the moment the UPS config is that way that it shutsdown down very early after the UPS switches to battery. When Battery goes below 95% or the rest of battery minutes go below 15min. So there should be more then enough capacity or runtime to shutdown the RaspiBlitz. The first goal is to have the UPS to keep the RaspiBlitz thru very short power outages and shut it down safely in case the power stays out longer then a very short time ... even if the UPS can still run for much more. That can be later optimized.

Kixunil, 2019-08-18:
That sounds reasonable for starters. I'd love to see it optimized later. Maybe just make the condition with those 15min? I guess 15 min should be plenty of time.

rootzoll, 2019-08-27:
I created the #735 issue for further optimization of this feature. For v1.3 this should be good.

closed 2019-09-03 by rootzoll