raspiblitz/raspiblitz #708: Experimental v1.3 RC2 image for the RPi4

openoms, 2019-08-05:
Made a rogue image after struggling to get Tor work with the v1.3 RC1 to prepare for the official v1.3 release:
Tested on my RPi4 well apart from RTL. Should work on an RPI3 too (did not try yet).

Download link: https://mega.nz/#F!AMc3RABR!EisZPKYKGywVh8OO3VVmZw

Check the integrity with:
`sha256sum raspiblitz-v1.3RC2-2019-08-05.img.gz`  
The result must be:  
e2aa397a42ff86c13b6a1282c5ec4d92abf93274fbcb82ed16016a071de47dfb  raspiblitz-v1.3RC2-2019-08-05.img.gz

Burn with [Etcher](https://www.balena.io/etcher/).
Log in with:
`ssh admin@RAPIBLITZ_IP`  
password: raspiblitz

Based on the Rasbian Buster with Desktop
https://downloads.raspberrypi.org/raspbian_latest

Bitcoin Core: 0.18.0
LND: 0.7.1

Built with:  
`wget https://raw.githubusercontent.com/openoms/raspiblitz/1.3RC2/build_sdcard.sh && sudo bash build_sdcard.sh 1.3RC2 openoms`

SDcard build log: https://gist.github.com/openoms/0d186e22fd4d3e22cfff34f26e1e7fae

Known issue:
* RTL does not work out of the box (manual install should be possible): https://github.com/ShahanaFarooqui/RTL/issues/173

Testing required if these are solved:
* Tor was not working on the previous 1.3 RC1 release: https://github.com/rootzoll/raspiblitz/issues/592
* RTL hangs on RPI idle: https://github.com/rootzoll/raspiblitz/issues/700
* Problems with initial setup on 1.3 rc1: https://github.com/rootzoll/raspiblitz/issues/704

fluidvoice, 2019-08-05:
What was changed in this build, just the version of Raspbian (Buster w/desktop)?

openoms, 2019-08-06:
@fluidvoice that was always the base image (Stretch w Desktop previously).
See the list of commits in the v1.3 branch  since the RC1 (2019-07-04).

openoms, 2019-08-06:
1.3RC2 on RPi4 4GB, official power, supply 500GB SSD Samsung Evo 860 started from fresh with SSD formatted to NTFS in Windows before start.
Syncing behind Tor from start. 
Looking good so far. Will report how long the IBD will take this time.
[image]

openoms, 2019-08-06:
> 1.3RC2 on RPi4 4GB, official power, supply 500GB SSD Samsung Evo 860 started from fresh with SSD formatted to NTFS in Windows before start.
> Syncing behind Tor from start.
> Looking good so far. Will report how long the IBD will take this time.

My RPi4 freezes the second time at ~8% IBD. Reformatted the SSD and burnt a new SD after the previous occasion. Reacts to ping, not to SSH. Any ideas?
[image]

openoms, 2019-08-06:
As it looks it can be the problem with how the RPi4 uses these USB-SATA adapters: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931

"These errors may also appear due to poor power quality or overloading the Pi's maximum 1.2A downstream USB port current, but if they persist when using a powered hub then they are genuine UAS issues."
If you have a powered USB hub or even a Y-cable use it!

Please share if you find a usable piece:
https://github.com/rootzoll/raspiblitz/issues/691#issuecomment-518845582

woeisme, 2019-08-09:
> My RPi4 freezes the second time at ~8% IBD. Reformatted the SSD and burnt a new SD after the previous occasion. Reacts to ping, not to SSH. Any ideas?
> [image]

I really recommend owning a USB TTL adaptor for figuring out what's going on when situations like this happen. This is the one I [own](https://www.adafruit.com/product/954)

openoms, 2019-08-13:
It is time to close this as the v1.3RC5 image is available:
https://twitter.com/rootzoll/status/1161238599045656576

closed 2019-08-13 by openoms