raspiblitz/raspiblitz #710: APC UPS Configure thru UI & and auto-reinstall after PowerShutdown

rootzoll, 2019-08-06:
Somehow after the APC UPS feature from #707 is making a emergency powershotdown after booting up the apcupsd service is "inactive" - blitz-ups.sh status giving a "n/a" and the complete ups feature needs to be reinstalled to get it working again with:
```
/home/admin/config.scripts/blitz-ups.sh off
/home/admin/config.scripts/blitz-ups.sh on apcusb
```
Thats OK for the experimental implementation within v1.3 but for the future this  should be fixed automatically on bootup also letting the user know that a power emergency shutdown was happening (maybe call the XXshutdown script from apcupsd with an extra parameter).

Also make it possible for the future to activate/set the UPS by the shell GUI from main/settings menu.

NilByte, 2020-12-07:
I wonder if it would be possible to make this functionality more generic.

https://github.com/networkupstools/nut/

A lot of different UPS seems to be supported.

https://networkupstools.org/stable-hcl.html

rootzoll, 2020-12-10:
@NilByte the NUT project a great find - I mark and an idea for v1.8 release. Also any experience report welcome of someone getting this running on the RaspiBlitz.

NilByte, 2020-12-10:
Sound great.

There is an article here referring to a Rasberry Pi package.

https://openschoolsolutions.org/shutdown-servers-case-power-failure%E2%80%8A-%E2%80%8Aups-nut-co/

`sudo apt-get install nut-client nut-server usbutils`

NilByte, 2020-12-14:
Inspiration for UI configuration.

http://rockstor.com/docs/ups-setup/ups_setup.html#rockstor-nut-configuration

https://github.com/rockstor/rockstor-core/blob/master/src/rockstor/system/nut.py

jayrockk, 2022-01-23:
I installed NUT as a client. Here are my personal notes. Maybe it is useful.

On the server (a Synology device in my case):

sudo cat /etc/ups/upsmon.conf
...
#MONITOR myups@localhost 1 upsmon pass master  (or slave)
...

On Raspiblitz:

sudo mkdir /etc/ups
sudo apt install nut-client nut-monitor

#Test connection
admin@raspberrypi:~ $ upsc ups@192.168.0.104  2>&1 | grep -v '^Init SSL'

#edit upsmon.conf
sudo nano /etc/nut/upsmon.conf
#add these two lines to the file #
MONITOR ups@<server_ip> 1 monuser secret slave
#…
SHUTDOWNCMD "/home/admin/config.scripts/blitz.shutdown.sh"

#edit nut.conf
sudo nano /etc/nut/nut.conf
#edit: mode=netclient

#fix permissions and make service permanent
cd /etc/systemd/system
sudo chown -R nut:nut /var/run/nut
sudo systemctl daemon-reload
sudo systemctl enable nut-monitor
sudo systemctl status nut-monitor

thephijo, 2022-02-01:
I also configured a UPS with nut for my Raspiblitz using an Eaton 3S. 
I mainly followed this explanation: https://wiki.ipfire.org/addons/nut/detailed
There are also some options mentioned that I did not make use of. 
After having everything set up as describes (worked without any issues) I changed some parameters to better match the Blitz:
In /etc/nut/upsmon.conf I replaced the SHUTDOWNCMD "/sbin/shutdown -h +0" with SHUTDOWNCMD /home/admin/config.scripts/blitz.shutdown.sh to shutdown the Blitz without killing LND etc. 
I also wanted to change the behaviour once the power is lost. I wanted the Blitz to be powered down much sooner than the default value. So I used upsrw -s battery.charge.low=90 -u admin -p mypasswd (mypasswd should obviously be replaced) to det the value for low battery to 90%. It now will shutdown the Blitz at 90% battery remaining giving it all the time in the world to power off safely and during a power outage it would not need to work anyways. 

To view a complete list of the UPS variables, both informational and modifiable: use upsc UPS (while UPS is the Name of you UPS). To determine the modifiable variables for your UPS, their current settings and their available setting values execute the command: upsrw UPS

You can now modify the variables you wish using a command similar to the following: upsrw -s battery.charge.low=20 -u admin -p admin UPS

Again that might help for individual configuration:
https://networkupstools.org/docs/user-manual.chunked/ar01s06.html
https://wiki.ipfire.org/addons/nut/detailed

To test everything you can plug you laptop into the ups (for draining the battery faster), plug the blitz not into the USP but some other power source, connect the USB-cable of the UPS to the blitz and disconnect the UPS from external power. So the battery will drain because of the Laptop, it will eventually reach low battery state and execute the shutdown command for the Blitz.

NilByte, 2022-02-01:
For troubshooting and making report of NUT settings and access rights http://rogerprice.org/NUT/nut-report can be very useful, maybe also for sharing working configurations.

```
#! /bin/bash -u
# Report openSUSE NUT configuration
# Remove comments, blank lines and passwords
# 2015-10-16 RP
# 2019-07-14 RP Added automatic detection of configuration file and executable directories
#               Fixed multiple password exposure.
#               Fixed CMDSCRIPT commented instance not ignored.
# 2019-07-18 RP Fixed passwords not detected for remote UPSs.
#               Added lsusb UPS units
# 2019-12-21 RP Added test for journald not installed
# 2020-08-07 RP Get report from a Synology NAS, Mac
VERSION="2020-08-07"

# Where do operating systems hide configuration files?
# Inspired by https://diktiosolutions.eu/en/synology/synology-ups-nut-en/
CFG=( "/etc/ups" "/etc/nut" "/usr/syno/etc/ups" "/sw/etc/nut" )        
# Where do operating systems hide NUT executables?
BIN=( "/usr/sbin" "/sbin" "/usr/syno/sbin" "/usr/syno/bin" "/sw/sbin" )

# Uses Bash arrays: see 
# https://www.tldp.org/LDP/Bash-Beginners-Guide/html/sect_10_02.html
# https://www.tldp.org/LDP/abs/html/arrays.html
# Set to -x to debug this fine piece of software
set +x

# Trap unexpected exits. See trap -l for list.
# Cleanup for normal exit is done at point of exit
trap 'RV=$? ; cleanup; exit $RV' HUP INT QUIT ILL TRAP ABRT BUS FPE SEGV PIPE ALRM TERM

# Cleanup on exit
function cleanup {
# Remove any remaining temporary files and file based variables
LTF=$( ls /tmp/$SCRIPT_NAME.* 2>/dev/null )         # No output or action if no temp files
if [[ $? -eq 0 ]]
then N=$( echo -e "$LTF" | wc -l )
     echo "Deleting $N /tmp/$SCRIPT_NAME.* temporary files ..."
     rm /tmp/$SCRIPT_NAME.*
fi
}

SCRIPT_NAME=$( basename "$0" )                      # Name of this Bash script
S="[[:space:]]"; G="[[:graph:]]"                    # REs for one or more spaces; one or more characters
RE_COM_BL="^#.*$|^$S*$"                             # RE matches comment or blank line
T=$( mktemp -p /tmp $SCRIPT_NAME.XXXXX ); : > $T    # Temporary output file, initially empty
PU=$( mktemp -p /tmp $SCRIPT_NAME.XXXXX ); : > $PU  # Temporary file for upsd.users passwords, initially empty
PM=$( mktemp -p /tmp $SCRIPT_NAME.XXXXX ); : > $PM  # Temporary file for upsmon.conf passwords, initially empty
W=$( mktemp -p /tmp $SCRIPT_NAME.XXXXX ); : > $W    # Temporary work file, initially empty
R="/tmp/NUT.report"                                 # T without passwords
echo -e "\n        Welcome to $SCRIPT_NAME       Version: $VERSION"

# Warn that non-root use will give incomplete report
if [[ $( id ) =~ uid=[0-9]+\(($G+)\) ]]
then USER=${BASH_REMATCH[1]}
else USER="martian"
fi
if [[ $USER == "root" ]]
then :
else echo -e "\nWARNING: You are calling this script as a non-root user $USER."
     echo -e "This will probably produce an incomplete report.\n"
fi

echo "Collecting NUT configuration ..."
echo -e "        NUT configuration  $( date --utc '+%Y-%m-%d %T %Z' )" >> $T

# Operating system
sleep 1; echo "Operating system ..."
echo -e "\n        ########### Operating system ###########" >> $T
[... 214 lines cut ...]
# Bow out
mv $T $R
if [[ $? -ne 0 ]]
then echo -e "\nERRROR: Unable to create report file $R\n"
     cleanup
     echo "Good-bye"
     exit 1
fi 
NCHAR=$( cat $R | wc -c ); NLINE=$( cat $R | wc -l )
NKB=$(( ($NCHAR+1023) / 1024 ))
sleep 1
echo ""
echo "I have created file \"$R\", $NLINE lines $NKB KBytes, with a summary of"
echo "your NUT configuration.   Passwords have been removed.  To submit this report"
echo "to the NUT mailing list, upload the file to a web server if you have access to"
echo "one, or to a file hoster such as filehosting.org, and post the URL they give"
echo "you, or attach a gzip compression of file \"$R\" to a post in the"
echo "mailing list."
echo ""
cleanup
echo "Good-bye"

exit
```
I think that the approach that would work for most UPS is to schedule an early shutdown based on ONBATT with a timer of maybe 300 secs since some UPS doesn't allow modifying the LOWBATT value through battery.charge.low.