raspiblitz/raspiblitz #714: read-error on swap device

wmpedersen, 2019-08-07:
Just installed 1.2 and switched to a proper 3A power supply (the Argon in favour of my 2A Samsung charger) and during sync I noticed tonight after running for a while that it was showing 'read-error on swap device' and that my external hard drive was off. This drive is USB only and pulls power from the Raspberry Pi, so perhaps it's an issue with the individual power supply.

rootzoll, 2019-08-07:
Are the error messages consistent or was it just a one time occurence?

wmpedersen, 2019-08-08:
With the hard drive off it was one after the other. Unfortunately I didn't catch it while it was beginning and the drive was still on.

fluidvoice, 2019-08-08:
> This drive is USB only and pulls power from the Raspberry Pi

which drive is it, and do you know what it's current draw is?

wmpedersen, 2019-08-08:
It's the Toshiba HDTB410XK3AA (The one in your photo). I'm afraid I don't know what the draw is, but it's been running since last without any messages on screen. At about  97~98% percent now.

wmpedersen, 2019-08-10:
It's been running steady without issue and has crossed 99% on blockchain sync. I'm going to chalk this issue up to power weirdness and close this for now.

closed 2019-08-10 by wmpedersen