raspiblitz/raspiblitz #715: [Feature Request] Clone HDD & Copy Blockchain

CryptoSigmund, 2019-08-07:
If you are already running a Raspiblitz but you wish to change your HDD to a new SSD or want to prepare for a hackday... 

Storage Menu:
Clone entire HDD to a new disk -> lets you switch to a complete new HDD/SSD
Just copy the Blockchain to a new disk -> copy your blockchain to a friends HDD/SSD

What do you think?

rootzoll, 2019-08-07:
> Clone entire HDD to a new disk -> lets you switch to a complete new HDD/SSD

Sounds like a feature for a v1.4 maybe to make it easier for people to move from HDDs to SSDs now that 500GB SSDs get cheaper and RP4 has USB3.

> Just copy the Blockchain to a new disk -> copy your blockchain to a friends HDD/SSD

Its a nice feature ... but not high on my prio list, because there is already the COPYSTATION mode described in the WORKSHOP.md that is offering that feature but in a even more bigger sheme to prepare and keep in sync multiple HDDs from one RaspiBlitz to prepare a workshop --> https://github.com/rootzoll/raspiblitz/blob/master/WORKSHOP.md#prepare-blockchain-copy-station

nyxnor, 2021-04-18:
This is done, pls close this.

closed 2021-04-19 by rootzoll