raspiblitz/raspiblitz #717: [Feature Request] Zap Desktop connect

CryptoSigmund, 2019-08-10:
It whould be nice if there was one more way for Zap to connect... and that is the desktop wallet:

/usr/local/gocode/bin $ lndconnect -i -j

this will generate a copy and paste text for the connection with the local ip to the fullnode. So no more unhealty store of the tls and macaroon on your pc. This whould also improve security of the wallet.

rootzoll, 2019-08-12:
Can be added to the EXPORT menu in a future version.

openoms, 2019-08-13:
for me the command was:
`/usr/local/gocode/bin/lndconnect -j`

The code showed my external public IP, which I have edited to my dynDNS and worked well.

CryptoSigmund, 2019-08-13:
the -i is for the internal ip and without it you get the external like openoms tested.

frennkie, 2019-10-03:
I'm not sure whether I exactly understand the problem/request. According to `lndconnect --help` there are these flags which appear relevant:

```
LndConnect:
  -i, --localip               Include local ip in QRCode
  -l, --localhost             Use 127.0.0.1 for ip
      --host=                 Use specific host name
  -p, --port=                 Use this port (default: 10009)
  -o, --image                 Output QRCode to file
```

The ZAP script (`97addMobileWalletLNDconnect.sh`) uses `--host`, `--port` and `--image`.

The host flag will default to the local IP but might be changed to a DynDNS or Reverse SSH value later in the script. 

1. You would like to have an additional option, right? 
2. Which should use a local IP, right? 
3. Are you looking for `127.0.0.1`?
4. Or something like `192.168.1.42`?

nyxnor, 2021-04-18:
lndconnect kinda ended. Pls close this.

closed 2021-04-19 by rootzoll