raspiblitz/raspiblitz #720: LND fails to start

osilloh, 2019-08-13:
debugs are at this link.

https://termbin.com/y274

Pi4 4g, original Pi power measely 15.3w (low power warning, proceeded with install anyway) with Fanshim, Seagate STEH1000100 1TB HDD USB3, Kuman LCD uninstalled because of power warning.

[image]

LND not ready 

[image]

rootzoll, 2019-08-13:
Is it correct, that you have TOR activated? The RC1 had still some problems with TOR and from the LND logs it seems that LND is nit able to speak with the TOR service/socket: Aug 13 05:26:02 RaspiBlitz lnd[29950]: dial tcp 127.0.0.1:9050: connect: connection refused

Today or tomorrow I will release the v1.3 RC5 - I tested TOR a lot woth that and it looks good. To reuse your already available blockchain with the new release: Go to terminal (CTRL+C), use command "repair" and choose "RESET HDD (but keep blockchain)". Then once the RaspBlitz is powered down ... burn the new RC image to the sd card and start again from the beginning.

I will also adapt the power levels for the power outlet.

fluidvoice, 2019-08-13:
> debugs are at this link.
> 
> LND not ready
> 

I'd like to see your Tor error, please run this and paste the link:
`sudo grep "tor[" /var/log/syslog | nc termbin.com 9999`

osilloh, 2019-08-19:
Yes, I have TOR activated. I have downloaded RC5, I shall give it a try and post any issues. thank you both for your help.

osilloh, 2019-08-20:
Success!

RC5 worked like a champ. Followed your instructions above and it came up. It still complains about low power as you can see. 

[image]

Now I just need to fund it and open some channels. Will try to build a BTCpay server as well.

[image]

closed 2019-08-22 by osilloh