raspiblitz/raspiblitz #724: [Help pls] Need help to recover funds 25% wallet 75% Channels. Version < 1.1

WebBlockChainDev, 2019-08-15:
Hi, i got a problem with my raspiblitz, i cant connect through ssh and cant connect through RTL.
I got access to the HDD and SD card though.
My display says i need to login with root buy my masterPW dont work, i tried with a direct keyboard on the raspberry pi,
What can i do here to backup my funds / btc in open channels + wallet?  

Previous Errors, now i dont even get that far anymore:
https://i.imgur.com/24qB1NO.jpg
https://imgur.com/YNuXFlc
https://i.imgur.com/n3WPjTE.png 
 
I m not sure how to recover my funds here now.
I got 25% of my funds in the wallet and 75% in channels.

Since i have a older version (i m not sure where to see which version), i dont have the following things to recover my funds:
v1.1 lnd.rescue.sh
v1.2 channel.backup

So i m not sure now what todo without losing funds. I read alot that people r able to get their wallet funds through their Seed but they dont get the channel funds back. Some even dont get the funds of the wallet balance what makes me worry now since my amount is not that small here.

Previous Issue: https://github.com/rootzoll/raspiblitz/issues/674

openoms, 2019-08-16:
Burn a new SDcard with the most recent version.

Booting with the new SD will recover all your funds if the data on the HDD is intact.

You can also boot your computer with a Linux USB (as suggested by @rootzoll before) and make a copy of the lnd directory on the HDD. That would allow to restore your funds if copied over to a working RaspiBlitz.

WebBlockChainDev, 2019-08-16:
I just tried as u mentiond to burn v1.2 on a new SD card.
After starting the raspberry pi with the new SD card and the old HDD and connecting through ssh i had to pick a new name for the raspiblitz and setup the passpords A and B.

After those steps it showed the following:
https://i.imgur.com/YQ9kSPn.png

It was unexpected to me so i did "sudo reboot" 
After reboot it showed this same as the old SD card where i´m not able anymore to login with my password
https://i.imgur.com/EJmsWRS.jpg

I repeated all steps until this message 
https://i.imgur.com/YQ9kSPn.png

Now i m at this point: https://i.imgur.com/j03aNmn.png

And i dont know where to go from here.

openoms, 2019-08-17:
It seems that your HDD is corrupted. 

There is a slight chance that it is because it does not get enough power.
Best would to try to attach it to PC booted with Linux to see if you can copy the lnd directory. 
Here is how you can make a bootable Ubuntu USB: https://tutorials.ubuntu.com/tutorial/tutorial-create-a-usb-stick-on-windows#0 
Test it only, you shouldn't modify your PC system at all apart from booting form USB. 
Alternatively you can try to use  Y-cable with external power or a powered USB hub to give exyra power to the HDD when you connect to the RaspberryPi.

There is a chance, that your lnd data is gone. 
In that case you should be able to get back the onchain funds with the seed.
There is an experimental method to attempt to get back the funds in channels as well following this guide: https://medium.com/@guggero/did-you-lose-funds-on-the-lightning-network-because-of-a-disk-crash-8971b6a92494

WebBlockChainDev, 2019-08-17:
I opened the Hdd with a software called "DiskInternals Linux" and yes i m able to backup the "lnd" folder.
But what i do with it?

openoms, 2019-08-18:
Good news. I recommend to run with the new SDcard to obtain a channel.backup with the new lnd version. 
Your HDD was corrupted (lucky that the channel information seems to be intact), make sure you have a backup off device as it is likely to happen again. You could benefit from continuous backup to  Dropbox as well:
 https://github.com/rootzoll/raspiblitz#b-dropbox-backup-target

Consider using a Y cable and extra power or a powered USB hub for you HDD as usually the problem lies with power supply issues to the HDD.

WebBlockChainDev, 2019-08-19:
Unfortunately an error appeared after 15% sync of the bitcoin blockchain.
https://i.imgur.com/JlcNtsS.png
Error log: https://termbin.com/cq1n

I was able to connect to RTL while syncing and see the total amount of wallet & channel funds in there.
After the error, i was able to get into the main menu but not be able to really do anything in there since the blocks werent synced.
After reboot i cant login anymore now and i have to newly setup the SD again.

Any idea what todo now? I just need to somehow close the channels to afterwards use my seed to save the funds. Can i somehow close the channels without syncing the blocks?

openoms, 2019-08-27:
Your data hard disk still does not seem to function properly. Did you change anything with the power (powered SB hub/ Y cable)?
If you are running in a closed case, remove the sides to avoid high temperatures too.

Once you get everything synced you will need to copy your backed up lnd directory in place of /mnt/hdd/lnd/:

* login with ssh to your RPi and stop lnd with:
`$ sudo systemctl stop lnd`

* copy the backed up lnd directory from your PC in place of `/mnt/hdd/lnd/`
You can do it with SCP from the terminal or use WINSCP as described here: https://stadicus.github.io/RaspiBolt/raspibolt_50_mainnet.html#copy-using-winscp

* when done restart your RPi from the terminal:
`sudo reboot`

When starts up again it should come up with your old funds and channels. Will need to use your old lnd wallet paasword (password C) to unlock the wallet.

WebBlockChainDev, 2019-09-04:
> Your data hard disk still does not seem to function properly. Did you change anything with the power (powered SB hub/ Y cable)?
> If you are running in a closed case, remove the sides to avoid high [...]

Thank you.
My node got synced now after copying the backup LND folder and i got my channels and funds back and i was able to close my open channels and send the funds to my normal wallet.

I still have to make a new setup again because some commands in the main menu like for example checking balance give out some errors and dont work.

closed 2019-09-04 by WebBlockChainDev