raspiblitz/raspiblitz #727: [Test] Power Supplys for RaspberryPi 4

CryptoSigmund, 2019-08-17:
In the last days I tested different models of power supplys and want to share you my results:

Testsetups:

1. Raspiblitz
Raspiblitz 1.3 rc5
Samsung 950 pro 1TB
ShimFan
LCD

2. Raspiblitz
Raspiblitz 1.3 rc5
Samsung 960 pro 1TB
passive cooler metal case
LCD

Power Supply:

1. Apple 61W USB-C Power Adapter MRW22ZM/A
passed all testsetups

2. Apple 29W USB-C Power Adapter MJ262Z/A
warnings on all testsetups

3. Anker 15W PowerPort C1 USB C
passed all testsetups

4. Anker 60W Anker PowerPort+
warnings with some criticals on all testsetups

5. Stotronics USB-C Power Supply - 5.1V 3A, RPI-18PFCA-05
passed all testsetups

6. Raspberry Pi 15.3W USB-C Power Supply
warnings with many criticals on all testsetups

So the best deal until know is the Stotronics which you can order from https://shop.pimoroni.com or the more expensive Anker PowerPort C1. 

If you have some other experiences, I whould be glad if you add them.

fluidvoice, 2019-08-18:
This was tested on Pi3 or Pi4? I did not find this Stontronics PS on Pimoroni or anywhere else.
Also if any of these PS's use USB cable instead of their own wire that should be listed too because the wire can make a big difference to drops in voltage.

CryptoSigmund, 2019-08-18:
All on Raspberry Pi4. The Stontronics is the alternative from Pimoroni to the original. You can see this at the order package https://shop.pimoroni.com/products/raspberry-pi-4

That power supplys that don't have a own cable, i used Anker Powerline II USB C to USB-C 2.0 cabel 1,8M length.

fluidvoice, 2019-08-19:
> All on Raspberry Pi4. The Stontronics is the alternative from Pimoroni to the original. You can see this at the order package https://shop.pimoroni.com/products/raspberry-pi-4
> 
> That power supplys that don't have a own cable, i used Anker Powerline II USB C to USB-C 2.0 cabel 1,8M length.

Ah OK good to know.  So you mean the "Worldwide" (universal) PS here [sold on Pimoroni here](https://shop.pimoroni.com/products/universal-usb-c-power-supply-5-1v-3a). 
Sontronics also known as "TT Electronics"

[image]

asle, 2019-08-23:
Raspiblitz
Raspiblitz 1.3 rc5
Seagate Expansion Portable 1TB (STEA1000400)
ShimFan
LCD

I get low power errors on the 
-Stontronics power supply
-Original Raspberry USB-C power supply

62.0'C  600/1500 MHz 0.8384V
--> Power CRITICAL detected
--> Power WARN detected

fluidvoice, 2019-08-23:
I commented about this on the Telegram channel.  This is most likely are result of this particular (unpowered) Seagate HDD. If you read reviews about this drive on Amazon and Newegg you will see other people, particularly using a notebook or a Pi saying it pulled too much power and needs a powered USB hub to work in their config.  
  Also, IMO Seagate drives have iffy QA and durability/longevity quality compared to WD, Toshiba, Samsung, etc.  I personally would stay away from Seagate drives except perhaps for less often used backup drives.

openoms, 2019-08-24:
We did see before that the Voltage is not the same on all the boards: https://github.com/rootzoll/raspiblitz/issues/673#issuecomment-507701909

I would suggest running the hardware test with everything connected and if there is a fault with all peripherals (LCD, HDD) disconnected too.
If there is a difference in the Volt values then it is likely to be a problem of the peripherals (likely the disk) drawing too much power.
If there is no difference I would suggest to carry on installing/syncing and if there are no further problems we should adapt the powertest to the lower numbers.

CryptoSigmund, 2019-08-30:
In the long term I discovered now that the:

Anker 15W PowerPort C1 USB C

now throws warnings and criticals - so no more recommendations on this one!

asle, 2019-08-30:
I have had nothing but problems with this setup. I have lost count of reinstallation of SD card, format HDD, different power supplies, but problems and more problems. Then a friend set up the exact parts like me, same Pi4/4GB, same SD card, same HDD (Seagate Expansion 1TB USB) and touchscreen. And official 3A USB-C power supply. Installation was a breeze and passed all tests. Copied over a downloaded blockchain just like I did. Everything is running fine. So I am returning the Pi4 to get a new one. Hopefully this will solve all the problems.

Xorell, 2019-11-14:
I am working en an RPI4 4GB, a 1TB MX500, shim fan and a LCD. I power my ingredients by use of the official raspberry PI supply. I consistently get the EXT4-fs errors during the download of the blockchain (after 5 to 30 minutes of downloading). I was reading many issues regarding the supply, but since i am working with the recommended hardware I would not expect issues like this. 

I tried to download without FAN or without LCD but any of these worked. Then i tried to attach a variable external supply to the PI (instead of the USB C supply), I've set the voltage to 5.1V ~ 5.2V and ever since the PI is flawlessly downloading the chain.

Do you think i am dealing with malfunctioning hardware or should i go for the Stontronics ?

rootzoll, 2019-11-15:
Thanks for your testing - I removed the Official Power Supply from most Shopping lists and recommending there now the Stontronics. I will keep the issue open for now for further testing and results .. but my get closed when no further reports comming until end of 2019.

thibaultmol, 2020-12-21:
Any updates on this testing? (changes with existing chargers. New chargers..?)