raspiblitz/raspiblitz #728: 1.3 RC5 display switch to sync screen

CryptoSigmund, 2019-08-18:
I have the behavior that at a timeframe from about 3-4 hours it switches a short period (about 20 seconds) to the blue screen with the blockchain and lightning % sync back and then to the normal screen. 

Does anyone encountered such behavior?

fluidvoice, 2019-08-19:
what is "the normal screen"?

CryptoSigmund, 2019-08-25:
Sry ;) (normal screen)

[image]

openoms, 2019-08-27:
> I have the behavior that at a timeframe from about 3-4 hours it switches a short period (about 20 seconds) to the blue screen with the blockchain and lightning % sync back and then to the normal screen.
> 
> Does anyone encountered such behavior?

This is normal behavior. When there is a new block propagated to your node it needs to verify the data and sync lnd as well.

CryptoSigmund, 2019-08-27:
Thank you for clearing this up, just curious about to turn this off?! If you have a node in a room where someone works/sleeps it's like disstracting if it switches to that blue screen and back to the "dashboard".

I think it whould be better if the blue screen is just for the IBD and then when a new block comes it is shown in the "dashboard". Otherwise the 100% there will never change?
[image]

openoms, 2019-08-27:
You could reduce the refresh rate of the screen to avoid catching the short syncing periods.
Could edit:
`sudo nano /home/admin/00infoBlitz.sh`
Increase the `sleep 5` value towards the end in the line 297.
`sleep 30` would result to refreshing only in every 30 seconds which would make it less likely to run into a syncing screen.

rootzoll, 2019-08-28:
Hopefully with a future version you can also use a epaper with no backlight .. that would be better to have it in your sleeping room anyway. I will close this issue for now.

closed 2019-08-28 by rootzoll