raspiblitz/raspiblitz #729: backup / restore procedure

woeisme, 2019-08-19:
Is there concise directions I'm missing about how to properly back up and restore important keys / pass phrases on Raspiblitz? 

There's a backup tab on RTL interface but very little information on what it does or where it goes. 

Perhaps a backup utility script to a USB device is a good solution?

fluidvoice, 2019-08-19:
Does this help? https://github.com/rootzoll/raspiblitz#backup-for-on-chain---channel-funds

woeisme, 2019-08-21:
> Does this help? https://github.com/rootzoll/raspiblitz#backup-for-on-chain---channel-funds

yes it does, but I think this should probably be added as a service to the main menu along with bitcoin wallet backup option.

fluidvoice, 2019-08-22:
Perhaps but it would need to be done with extreme care and ideally with user interaction.  If something goes wrong the user needs to catch it else funds can be lost.  So maybe the scripts can automate some things but prompt the user to verify results?  not sure.

rootzoll, 2019-08-28:
The external backup options for the channel.backup file were just on command line to test them first. But they look good so far, so I consider adding them to the main menu for a release after v1.3.

woeisme, 2019-09-06:
> 
> 
> The external backup options for the channel.backup file were just on command line to test them first. But they look good so far, so I consider adding them to the main menu for a release after v1.3.

maybe some option like:

> plug in a pendrive to backup all important wallet files

and user can insert a pendrive for that purpose.

rootzoll, 2019-09-09:
@woeisme havin a pen drive for constant backup for all your LND data (as part of a RAID with the HDD) is already part of another issue #329 ... the "external" in the issue is the idea of having an "off-location" backup.

rootzoll, 2020-03-21:
For version 1.5 RC1 the remote backup of the static channel backup file will be available from the service menu. Tested. Closing issue.

closed 2020-03-21 by rootzoll