raspiblitz/raspiblitz #730: Zap install, GO download link incorrect on RPi 4

Timbo925, 2019-08-21:
Receiving the following error while trying to install zap mobile service. It seems like the download link is incorrect for the GO framework. This is happening on the RC release v1.3 and a Raspberry Pi 4.

Running `umame --m` returns `ARMv7l`, 

Managed to get it working by installing version go1.12.9.linux-armv6l.tar.gz. Then setting the PATH correctly like in the following script. Then running the setup again. 

````./97addMobileWalletLNDconnect.sh: line 36: [: -eq: unary operator expected
./97addMobileWalletLNDconnect.sh: line 39: [: -eq: unary operator expected
./97addMobileWalletLNDconnect.sh: line 42: [: -eq: unary operator expected
./97addMobileWalletLNDconnect.sh: line 45: [: -eq: unary operator expected
*** Installing Go v1.12.6 for  ***
--2019-08-21 21:14:38--  https://dl.google.com/go/go1.12.6.linux-.tar.gz
Resolving dl.google.com (dl.google.com)... 2a00:1450:400e:80d::200e, 172.217.20.110
Connecting to dl.google.com (dl.google.com)|2a00:1450:400e:80d::200e|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-08-21 21:14:38 ERROR 404: Not Found.````

openoms, 2019-08-22:
Thank you for reporting. You have caught a typo! PR is in to fix for the next release.

rootzoll, 2019-08-27:
Merged the PR to fix this. On final testing for v1.3 release.

openoms, 2019-09-01:
Having trouble here , Go does not have an ARMv7l only ARMv6l -> will need to change that back.

Also we are checking for Go 1.11 as that is in the outdated LND docs. 
The warning about :
```
### Check Framework: GO ###                                                                                                                                                                                                                                                                                                                                                                                                        
WARNING: You work with a untested version of GO - should be 1.11 .. trying to continue           
```
could be removed as it is confusing.

rootzoll, 2019-09-04:
Ok tested. Go works good now.

closed 2019-09-04 by rootzoll