raspiblitz/raspiblitz #740: LND isn't starting anymore.

pg024, 2019-08-28:
Hello, LND is not running anymore on a rasp pi 3. It has been running smoothly this year, then I shut down the whole PI for 2 months. 
Now I have turned the pi 3 on again but the lnd for some reasons fails after a Restart loop.

https://termbin.com/fl8f

pg024, 2019-08-28:
I noticed that with TOR disabled LND is able to start.

pg024, 2019-08-28:
But TOR itself does not look connected.

pg024, 2019-08-28:
TOR logs.

https://termbin.com/zcbl

rootzoll, 2019-08-28:
The v1.2 had some problems with TOR - those should be fixed with v1.3 that is not final yet, but you can try the v1.3 Release Candidate 5 see for download here: https://github.com/rootzoll/raspiblitz/tree/v1.3#installing-the-software

fluidvoice, 2019-08-29:
I found some docs and issues around a "dormant" feature of Tor which could be related.
If Tor has a state when not used for a while it needs some unique command/action to change this state?? It also looks like there have been changes related to this in the most recent Tor releases (so potentially affecting RB v1.3? : https://blog.torproject.org/new-release-tor-0401-alpha

[image]

frennkie, 2019-09-26:
ran into this too... fix my specific situation required: `sudo chown bitcoin:bitcoin /run/tor/` and remove `User bitcoin` and `PortForwarding 1` from `/etc/tor/torrc`

I'll monitor further..

closed 2019-11-16 by rootzoll