raspiblitz/raspiblitz #741: LND Wallet Locked (LND restart loop)

wmpedersen, 2019-08-29:
I'm having an issue where I have powered up the 'blitz and it starts syncing after unlock, but after a few minutes I'm prompted on the screen that LND has restarted and to unlock my wallet again. Once complete it goes back to syncing until it shows the restart/unlock screen again.

I've enabled auto-unlock as a potential workaround, but I have a feeling it's still going to be restarting for some reason.

```
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 11:23:26 up  1:03,  2 users,  load average: 2.73, 2.82, 2.69

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-08-29 10:23:43 BST; 59min ago
  Process: 2133 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 2126 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 2193 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─2193 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitco…id
Aug 29 10:23:42 zeus systemd[1]: Starting Bitcoin daemon...
Aug 29 10:23:43 zeus systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sat 2019-08-24 20:17:01 BST, end at Thu 2019-08-29 11:23:26 BST. --
Aug 29 10:23:42 zeus systemd[1]: Starting Bitcoin daemon...
Aug 29 10:23:43 zeus systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-08-29T10:21:32Z UpdateTip: new best=000000000000000000058cce1a08e4b3bb1073673922cf6535f5ca2df91e5c66 height=591597 version=0x20000000 log2_work=91.001909 tx=448323056 date='2019-08-24T22:01:24Z' progress=0.997917 cache=61.5MiB(542373txo)
2019-08-29T10:21:49Z UpdateTip: new best=0000000000000000000ba48d67829de1f82f7b9a98eaf4556b016822fdf2c701 height=591598 version=0x20400000 log2_work=91.001934 tx=448326405 date='2019-08-24T22:31:00Z' progress=0.997926 cache=62.1MiB(547679txo)
2019-08-29T10:21:49Z Pre-allocating up to position 0x5000000 in blk01769.dat
2019-08-29T10:22:07Z Pre-allocating up to position 0x900000 in rev01769.dat
2019-08-29T10:22:07Z UpdateTip: new best=0000000000000000001564d25432fce9cce5f732f47bfb891ab9776e3bdc20eb height=591599 version=0x3fffe000 log2_work=91.001959 tx=448329635 date='2019-08-24T22:43:41Z' progress=0.997930 cache=62.5MiB(551358txo)
2019-08-29T10:22:27Z UpdateTip: new best=000000000000000000031718f0c7a0fd2eeda0faf71725b74f5abc401c6e611f height=591600 version=0x20400000 log2_work=91.001985 tx=448333008 date='2019-08-24T23:09:20Z' progress=0.997938 cache=63.1MiB(557375txo)
2019-08-29T10:22:27Z New outbound peer connected: version: 70015, blocks=592265, peer=135
2019-08-29T10:22:32Z UpdateTip: new best=00000000000000000014d9cb22510d9be20c64bf06694593e042bb9453d5cceb height=591601 version=0x20000000 log2_work=91.00201 tx=448334574 date='2019-08-24T23:09:43Z' progress=0.997938 cache=63.1MiB(557809txo)
2019-08-29T10:22:35Z UpdateTip: new best=0000000000000000000d0c069b4edb29854624eb77ef0b30dea2e5d75d891c67 height=591602 version=0x20000000 log2_work=91.002036 tx=448335249 date='2019-08-24T23:12:53Z' progress=0.997939 cache=63.2MiB(558569txo)
2019-08-29T10:22:35Z Pre-allocating up to position 0x6000000 in blk01769.dat
2019-08-29T10:22:37Z UpdateTip: new best=00000000000000000000f0ab2ae2ee6368821664e41b4367bf38bec4ab9c4d52 height=591603 version=0x20000000 log2_work=91.002061 tx=448335339 date='2019-08-24T23:13:25Z' progress=0.997939 cache=63.3MiB(558810txo)
2019-08-29T10:22:45Z UpdateTip: new best=000000000000000000180298d69a9153e2cd06ec24fb720ebaa4af22dccab0bd height=591604 version=0x20000000 log2_work=91.002087 tx=448337597 date='2019-08-24T23:23:01Z' progress=0.997942 cache=63.4MiB(560441txo)
2019-08-29T10:22:48Z UpdateTip: new best=00000000000000000000a1b0bfd836dfd4bf6a37189d4176c7e21e6be69134d4 height=591605 version=0x20000000 log2_work=91.002112 tx=448338188 date='2019-08-24T23:26:44Z' progress=0.997943 cache=63.5MiB(561058txo)
2019-08-29T10:22:52Z UpdateTip: new best=000000000000000000107c2e276c46d7ac42770d7a7a2329d018ad672ed9484c height=591606 version=0x20000000 log2_work=91.002138 tx=448339272 date='2019-08-24T23:31:34Z' progress=0.997945 cache=63.6MiB(561657txo)
2019-08-29T10:22:52Z Pre-allocating up to position 0x7000000 in blk01769.dat
2019-08-29T10:22:55Z UpdateTip: new best=00000000000000000000735fed83a08672e5c561a33988418823d5ae3c71e427 height=591607 version=0x3fffe000 log2_work=91.002163 tx=448340096 date='2019-08-24T23:35:32Z' progress=0.997946 cache=63.6MiB(562244txo)
2019-08-29T10:23:19Z UpdateTip: new best=00000000000000000006331df4e77bcb6f6d08f08438a7c643851960ff4c9e09 height=591608 version=0x3fffe000 log2_work=91.002189 tx=448342073 date='2019-08-24T23:45:34Z' progress=0.997949 cache=63.9MiB(565225txo)
2019-08-29T10:23:24Z UpdateTip: new best=000000000000000000179bcb104a42daed9a24f7a8c5cbf2384675818ab6934d height=591609 version=0x20000000 log2_work=91.002214 tx=448343066 date='2019-08-24T23:50:38Z' progress=0.997951 cache=64.0MiB(566182txo)
2019-08-29T10:23:24Z UpdateTip: new best=000000000000000000160b75ff10fb93ccaf9c615881eefb13463ad70ff75f6f height=591610 version=0x20000000 log2_work=91.002239 tx=448343125 date='2019-08-24T23:51:14Z' progress=0.997951 cache=64.1MiB(566255txo)
2019-08-29T10:23:24Z UpdateTip: new best=0000000000000000001978b3c9c4cc5b9b425d8ecd1f11effa05e695599b13bc height=591611 version=0x20000000 log2_work=91.002265 tx=448343211 date='2019-08-24T23:51:23Z' progress=0.997951 cache=64.1MiB(566346txo)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-08-29 11:18:40 BST; 4min 45s ago
  Process: 26264 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 26271 (lnd)
   CGroup: /system.slice/lnd.service
           └─26271 /usr/local/bin/lnd --externalip=198.48.187.228:9735

Aug 29 11:18:40 zeus systemd[1]: Starting LND Lightning Daemon...
Aug 29 11:18:40 zeus systemd[1]: Started LND Lightning Daemon.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Sat 2019-08-24 20:17:01 BST, end at Thu 2019-08-29 11:23:26 BST. --
Aug 29 11:10:28 zeus systemd[1]: Starting LND Lightning Daemon...
Aug 29 11:10:28 zeus systemd[1]: Started LND Lightning Daemon.
[... 54 lines cut ...]
powerMIN='1200000 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='4400 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.42'
tempCelsius='66.6'
uptime=3802
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=591613
initialSync=1
syncProgress=99.80
startcountLightning=4
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
scriptRuntime=30
```

rootzoll, 2019-09-01:
LND is restarting and locking wallet again - there is a panic log:
```
Aug 29 11:17:40 zeus lnd[21428]: panic: page 7846 already freed
Aug 29 11:17:40 zeus lnd[21428]: goroutine 1 [running]:
Aug 29 11:17:40 zeus lnd[21428]: github.com/coreos/bbolt.(*freelist).free(0x1dc7680, 0x807cc, 0x0, 0x64b25000)
Aug 29 11:17:40 zeus systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Aug 29 11:17:40 zeus systemd[1]: lnd.service: Unit entered failed state.
```

A lot of panic bug were fixed with the new 0.7.x versions ... maybe try the quick LND update withthe scripts of openoms: https://github.com/openoms/lightning-node-management/blob/master/lnd.updates/README.md#update-lnd-to-v071-beta

or wait for the final release of RaspiBlitz v1.3 hopefully by the end of week.

wmpedersen, 2019-09-02:
I ran the update script for 0.7.1, but no joy. Still rebooting

gehrlein, 2019-09-07:
I had this issue since v1.2 - I took my node offline until v1.3 was officially released. The error still persists after upgrade. I even ran the script again to update to v0.7.1 (but as it is already in v1.3 it simply reinstalled LND, I guess). Still the error persists.

rootzoll, 2019-09-09:
I would recommend to see if the LND dev Slack channel has more info on that issue: https://dev.lightning.community/

Another way would be to try to start with a fresh v1.3 LND wallet and use the SEED + channel.backup file to recover funds and then build up channels fresh.

AngelicInsight, 2019-09-27:
@gehrlein were you able to find a solution to this issue? I also have what seems to be an LND restart loop. Rescue file was not successful. 

I'll note that I had v1.3 working properly after a successful upgrade. I was attempting to activate LND Auto-Unlock and disable channel autopilot in the services menu. After restart, I've been stuck.

gehrlein, 2019-09-27:
@AngelicInsight No. Even though I made a clean install of v1.3 and resetted the HDD (without deleting the blockchain). Pretty frustrating as I do not want to reload the whole chain again.

AngelicInsight, 2019-09-27:
@gehrlein I’ve tried even deleting the blockchain. No luck.

gehrlein, 2019-09-27:
@AngelicInsight But this would basically be a clean install? How can this then be not working?

wmpedersen, 2019-09-27:
I also upgraded to 1.3 when I saw the torrent available but the issue persists.

wmpedersen, 2019-10-06:
I've got it to work. I had to wipe the HDD and copy blocks and chainstate to it from another computer and do the set up all over again but the node has synced and no loops anymore.

closed 2019-11-20 by rootzoll