raspiblitz/raspiblitz #742: UNKNOWN STATE error while testing v0.13

pxsocs, 2019-08-29:
While testing v0.13, the setup process stops and sends the following message: 

```text
UNKOWN STATE - there is blockain data config, but blockchain service is not running
It seems that something went wrong during sync/download/copy of the blockchain.
Or something with the config is not correct.
Sometimes a reboot helps --> sudo shutdown -r now
```

Tried rebooting but no luck.
Full message details below.

```text
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.3
chainnetwork: bitcoin / main
 17:17:47 up 13 min,  3 users,  load average: 2.50, 1.84, 0.97

*** BLOCKCHAIN SYSTEMD STATUS ***
Unit bitcoind.service could not be found.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2019-08-29 17:04:00 BST, end at Thu 2019-08-29 17:17:47 BST. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-08-29T16:16:08Z dnsseed thread start
2019-08-29T16:16:08Z opencon thread start
2019-08-29T16:16:08Z addcon thread start
2019-08-29T16:16:08Z msghand thread start
2019-08-29T16:16:08Z init message: Done loading
2019-08-29T16:16:08Z Cannot create socket for fno4aakpl6sg6y47.onion:8333: unsupported network
2019-08-29T16:16:08Z Cannot create socket for toguvy5upyuctudx.onion:8333: unsupported network
2019-08-29T16:16:09Z Cannot create socket for ndndword5lpb7eex.onion:8333: unsupported network
2019-08-29T16:16:09Z Cannot create socket for 6m2iqgnqjxh7ulyk.onion:8333: unsupported network
2019-08-29T16:16:10Z Cannot create socket for 5tuxetn7tar3q5kp.onion:8333: unsupported network
2019-08-29T16:16:19Z Loading addresses from DNS seeds (could take a while)
2019-08-29T16:16:23Z 280 addresses found from DNS seeds
2019-08-29T16:16:23Z dnsseed thread exit
2019-08-29T16:17:05Z UpdateTip: new best=00000000000000000016810e9c7a0f404a2f4f1df959347c49f6eb472afd93e8 height=591886 version=0x20c00000 log2_work=91.009246 tx=448909992 date='2019-08-26T21:03:06Z' progress=0.998017 cache=1.4MiB(12316txo)
2019-08-29T16:17:10Z Cannot create socket for fno4aakpl6sg6y47.onion:8333: unsupported network
2019-08-29T16:17:11Z Cannot create socket for toguvy5upyuctudx.onion:8333: unsupported network
2019-08-29T16:17:11Z Cannot create socket for ndndword5lpb7eex.onion:8333: unsupported network
2019-08-29T16:17:12Z Cannot create socket for 6m2iqgnqjxh7ulyk.onion:8333: unsupported network
2019-08-29T16:17:12Z Cannot create socket for 5tuxetn7tar3q5kp.onion:8333: unsupported network
2019-08-29T16:17:46Z ThreadRPCServer incorrect password attempt from 127.0.0.1:58622

*** LND SYSTEMD STATUS ***
Unit lnd.service could not be found.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Thu 2019-08-29 17:04:00 BST, end at Thu 2019-08-29 17:17:48 BST. --
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory

*** LAST 20 RTL LOGS ***
-- Logs begin at Thu 2019-08-29 17:04:00 BST, end at Thu 2019-08-29 17:17:48 BST. --
-- No entries --

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
raspberryPi=3
powerFAIL=0
powerWARN=0
powerMIN='1231300 microVolt'
tempFAIL=7
tempWARN=15
tempMAX='7520 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.106'
tempCelsius='65.5'
uptime=823
upsStatus='OFF'
startcountBlockchain=0
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

UNKOWN STATE - there is blockain data config, but blockchain service is not running
It seems that something went wrong during sync/download/copy of the blockchain.
Or something with the config is not correct.
Sometimes a reboot helps --> sudo shutdown -r now
```

openoms, 2019-08-31:
Hi. Is that a RPi3? As I understand you have used the HDD of a previously wunning setup and just wrote  anew SD.

It is either that the data on the HDD is not complete/corrupted or the restore process did not go through properly.
If you have changed something in your hardware config there could be a power supply issue.

If there is bitcoin and lnd data on the HDD the setup process should restore the previous lnd wallet only making you to change the PASSWORD_A on the first login. 

If you think there is no hardware error/disk corruption could try again with a freshly burnt SD card to trigger the restore process again.

pxsocs, 2019-09-02:
It is a RPi3. 
__Update__: I've downloaded v1.2 again and it worked. Then I tried the 1.3 and ended up with the same error as before.
For some reason the upgrade is not completing. 
Will try a couple of things tomorrow and report back but for now I've downgraded to 1.2.

blackjid, 2019-09-06:
I have the same problem

pxsocs, 2019-09-06:
Ultimately, the only thing that worked for me was to restore LND:
```bash
./98repairMenu.sh
```
MAKE SURE to have your seed (passphrase if used) and channel backups saved!

nyxnor, 2021-04-18:
Pls close this. The cleanup must be complete.

closed 2021-04-19 by rootzoll