raspiblitz/raspiblitz #743: Trouble restoring from backup file with 1.3rc5

baebb, 2019-08-30:
Trying to use my backup file from a 1.2 node with rc5

Went through node setup and LND [didn't use it], then ran this:

`/home/admin/config.scripts/lnd.rescue.sh restore`

I scp'ed over the rescue file in another window then continued with the process:

`OK -> found file to restore: /home/admin/lnd-rescue-0f87bb985a047dcac24129cbdc5462c3.tar.gz
OK -> checksum looks good: 0f87bb985a047dcac24129cbdc5462c3
Stopping lnd...
OK

Cleaning LND data ...
OK

Restoring LND data from /home/admin/lnd-rescue-0f87bb985a047dcac24129cbdc5462c3.tar.gz ...
OK

Starting lnd...
OK

DONE - please check if LND starts up correctly with restored state and funds.
Keep in mind that some channels maybe forced closed in the meanwhile.

*** Check Basic Config ***
*** Check bitcoin Running ***
*** Check bitcoin Responding *** (can take a while)
OK - chainnetwork is working

*** Check LND Config ***
Creating LND config ...
OK - LND config written

*** Init LND Service & Start ***
TOR was not selected
Created symlink /etc/systemd/system/multi-user.target.wants/lnd.service → /etc/systemd/system/lnd.service.

waiting .
waiting ..
waiting ...
`
It starts back up, then I get presented with:

[image]

Tried a few times but same results. What should I try next?

openoms, 2019-08-31:
The problem seems to be that you are using the restore script before your node has been set up.

Did you try to restore your backup by folowing the second option on your screenshot? It will ask if you have a backup file and guide you through.

baebb, 2019-09-01:
@openoms I follow the 2nd option and the cycle just repeats itself

openoms, 2019-09-01:
To make sure there is no power or hardware problem I would suggest to carry on with a new wallet then.

Once everything is synced 100% you can run the restore script
`/home/admin/config.scripts/lnd.rescue.sh restore`
to change to your old wallet.

baebb, 2019-09-01:
> Once everything is synced 100% you can run the restore script
/home/admin/config.scripts/lnd.rescue.sh restore
to change to your old wallet.

@openoms this is what I tried initially

openoms, 2019-09-01:
it probably means that the error is in the backup file. The hash is good, so file itself is not corrupted.
Was it a backup from a working node?

You can try uncompressing it manually too on your PC to check if it is good inside. It should be a copy of your old mnt/hdd/lnd directory. 

You can try again to scp in place of the /mnt/hdd/lnd directory of your node or try the lnd desktop app to restore from it: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-move-funds--channels-from-raspiblitz-to-lnd-lightning-desktop-app

baebb, 2019-09-02:
thanks @openoms , I will try that. The node was having low-power issues and not booting so I began the rescue process. Could the old node's LND auto-unlock configuration be causing an issue?

openoms, 2019-09-02:
No, it has nothing to do with the autounlock. With low power problems is probably a data corruption.
If you don`t succeed restoring the directory you should resort to restoring from seed and channel.backup which will end up closing your channels, but you get the funds (minus closing fees): https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#2-recover-from-wallet-seed

baebb, 2019-09-05:
@openoms sovled :) Thanks for your help

closed 2019-09-05 by baebb