raspiblitz/raspiblitz #744: USA parts list: power supply warnings

jarheadius, 2019-08-31:
Raspberry pi 3b+ running fine with warnings Argon ONE power supply isn’t good enough. 

Can someone recommend a more stable PSU?

rootzoll, 2019-09-01:
I am not from the US, so I was never able to test the Agron ONE. This power supply promises a good performance for the RP3: https://www.amazon.com/Raspberry-Micro-USB-Charger-Raspberry-Pi-Adapter/dp/B00L88M8TE

If you can test it and let us know that would be great. If its not giving warinings on the stress test REPAIR->HARDWARE I will use it for future shopping lists. Thx.

jarheadius, 2019-09-01:
Thanks I will purchase and provide feedback.

LukeParker, 2019-09-02:
That looks nice... Looking forward to jarheadius' feedback.
The one I bought works most of the time, but can't really keep the juice up over long periods of time: https://www.amazon.com/gp/product/B07FY9G348/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

After it's been running a couple of days I always get the "Weak power supply detected" message, and the hardware test shows it's erring on the weak side. It only had to power off once, and I didn't lose my channels so it's not the worst problem to have. I do love the in-line power switch on this model though.

jarheadius, 2019-09-20:
New PSU is being used. It’s taken a week to resync with lightning so I hope to have update soon. Is this normal to wait so long to sync after only unplugging for 15 minutes?

LukeParker, 2019-09-21:
Is it the sync where the blockchain is 100% but the LND % is actually counting backwards slowly? Try unplugging it and restarting the sync once or twice more. It usually take me two reboots, and one time a 3rd reboot to get past that problem. (Other tickets here talk about it.)

jarheadius, 2019-09-22:
That’s exactly what it doing.

Thanks I will check the other thread.

jarheadius, 2019-10-03:
New PSU tested and works perfectly. Thanks for the help.

closed 2019-10-03 by jarheadius