raspiblitz/raspiblitz #745: Test: Update from 1.2 to 1.3

rootzoll, 2019-09-01:
The update should be quite simple - you dont need to close any channels: 

- Best is to get a second 16GB or 32GB sd card (like in shopping list), download the [v1.3 image](https://github.com/rootzoll/raspiblitz#installing-the-software) from the github (once v1.3 is oficially released) and write it to the new sd card with a tool like [balena etcher](https://www.balena.io/etcher/). 
- In main menu of you RaspiBlitz choose update (if you have v1.3 image maybe choose `update anyway`) and follow the dialogs.
- Once your RaspiBlitz has shutdown, exchange the sd card and boot up again.
- The RaspiBlitz goes now thru a recover/update phase - this may take some time. Once thats done, login once per SSH and use the password `raspiblitz` and set a new password A (can be your old one or a new one).

After the final reboot your RaspiBlitz should be ready running the new version v1.3. Take good care about the old sd card, it contains a backup of the LND data with your private key and channel.backup file.

rootzoll, 2019-09-01:
TODO: Also test updating a v1.2 running TOR - to check if its possible to reproduce: #742

rootzoll, 2019-09-04:
Update Test with TOR worked.

rootzoll, 2019-09-04:
Also second test worked. v1.3 is ready to go.

closed 2019-09-04 by rootzoll

woeisme, 2019-09-12:
just a quick note that might help other users, was able to successfully use menu update to go from 1.3RC3 -> 1.3 tonight.