raspiblitz/raspiblitz #748: Unable to connect 8332 bitcoind RPC from LAN

openoms, 2019-09-03:
Trying to achieve to be able to communicate with my bitcoind with RPC, but not able.
Anyone succeded to do this on a RaspiBlitz? I have tried two nodes running 1.3RC2 and 1.3RC5 so far.

bitcoin.conf was modified to: 
```
sudo cat /mnt/hdd/bitcoin/bitcoin.conf 
# bitcoind configuration

# mainnet/testnet
testnet=0

# Bitcoind options
server=1
daemon=1
txindex=1
# disablewallet=1
listen=1

# Connection settings
rpcuser=raspibolt
rpcpassword=xxxxxxx
rpcport=8332

rpcallowip=0.0.0.0/0
rpcallowip=192.168.1.244
# rpcallowip=127.0.0.1
# rpcbind=127.0.0.1:8332
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=600
maxorphantx=10
maxmempool=300
maxconnections=40
maxuploadtarget=5000

datadir=/mnt/hdd/bitcoin
# onlynet=onion
addnode=fno4aakpl6sg6y47.onion
addnode=toguvy5upyuctudx.onion
addnode=ndndword5lpb7eex.onion
addnode=6m2iqgnqjxh7ulyk.onion
addnode=5tuxetn7tar3q5kp.onion
```

ufw configured:
```
ufw allow from 192.168.1.0/24 to any port 8332 comment 'allow RPC from local LAN'
ufw allow from 192.168.1.0/24 to any port 28333 comment 'allow RPC from local LAN'
```
then disabled.

Trying to make BTCpayserver connect to the RaspiBlitz, but the using with others too
* Samourai -> Settings -> Transactions -> Set trusted node too, but no cummunication happening (Trusted node not responding, Please check installation)
* telnet from an other machine on the LAN: 
```
$ telnet 192.168.1.205 8332
Trying 192.168.1.205...
telnet: Unable to connect to remote host: Connection timed out
```
on the RaspiBlitz: 
```
$ sudo lsof -i:8332
COMMAND    PID    USER   FD   TYPE   DEVICE SIZE/OFF NODE NAME
bitcoind 21334 bitcoin    9u  IPv6 43194284      0t0  TCP localhost:8332 (LISTEN)
bitcoind 21334 bitcoin   10u  IPv4 43194286      0t0  TCP localhost:8332 (LISTEN)

```

Any ideas why the port 8332 cannot be accessed?

openoms, 2019-09-03:
Got this now: 
 it works with` rpcbind=0.0.0.0` 

https://github.com/bitcoin/bitcoin/blob/master/doc/release-notes/release-notes-0.18.0.md#configuration-option-changes
> The rpcallowip option can no longer be used to automatically listen on all network interfaces. Instead, the rpcbind parameter must be used to specify the IP addresses to listen on. Listening for RPC commands over a public network connection is insecure and should be disabled, so a warning is now printed if a user selects such a configuration

closed 2019-09-03 by openoms

Lupus-Pluvia, 2019-09-03:
how can I change that for me it is 127.0.0.1 would also replace it with 0.0.0.0

openoms, 2019-09-04:
@Lupus-Pluvia what are you trying to achieve?

The rpcallowip= and rpcbind= needs to be the same to allow external RPC calls.
You would be best to use the LAN address of your external machine neeeding RPC access.

You can edit your bitcoin conf with
`sudo nano /mnt/hdd/bitcoin/bitcoin.conf`
save and restart bitcoind afterwards.

kilrau, 2019-12-13:
Goal: connect a LND running on my computer to bitcoind running on my raspiblitz pi4. Both are in the same ipv4 subnet (192.168.1.0/24).

This is the only way I could make it work:
```
rpcport=8332
rpcallowip=192.168.1.0/24
rpcbind=0.0.0.0
zmqpubrawblock=tcp://0.0.0.0:28332
zmqpubrawtx=tcp://0.0.0.0:28333
```
and configured UFW to allow local LAN connections for 8332, 28332, 28333.

Question: is there any chance that this could expose 8332, 28332 or 28333 to the public somehow, e.g. via TOR? Is there a more restrictive (not 0.0.0.0) way to get this done?

nymbuzcumulus, 2024-09-01:
So... does anyone got to run their asics with bggminer and btc core?? need help !