raspiblitz/raspiblitz #749: how can I connect my node with my 2 node

Lupus-Pluvia, 2019-09-03:
my 1 node  and 2 node  connect with each other despite port change
!Screenshot_20190903-174129_Chrome](https://user-images.githubusercontent.com/49962031/64188174-2d1d4f80-ce72-11e9-96a4-b375a9cba867.jpg)
and are displayed with an additional IP node 1 has port 9777 node 2 has port 9735
[image]

rootzoll, 2019-09-09:
I dont understand the problem? So you have two nodes running from the same public IP and want them to connect?

Lupus-Pluvia, 2019-09-10:
No, I want you to argue separately

rootzoll, 2019-09-17:
Sorry I still dont get a clear picture of your issue. I close this for now. If you write a more detailed description we can open the issue again.

closed 2019-09-17 by rootzoll