raspiblitz/raspiblitz #751: Connecting Electrum wallet to my RaspiBlitz

GiganteX, 2019-09-04:
So in Electrum there's a "Network" dialog, in the "Server" tab i de-select the "Select server automatically and instead I enter the IP of the RaspiBiltz in the "Server:" address editbox. And the port is the default 50002. I kind of expected that to just work, but no. It just will not connect. Am i in the wrong place, doing the wrong thing? Or is there some configuration settings on the RaspiBlitz that I need to do?

openoms, 2019-09-05:
For that to work you have to install an Electrum server. Two options:
* Electrum Personal Server: https://stadicus.github.io/RaspiBolt/raspibolt_60_bonus.html#electrum-personal-server (more suitable the RPi3)
* Electrs for a full server for devices with more RAM : https://github.com/openoms/bitcoin-tutorials/blob/master/electrs/README.md

GiganteX, 2019-09-05:
That is very strange to me. Electrum does not communicate with bitcoin nodes, only with Electrum servers?

openoms, 2019-09-05:
Use an Electrum Server of your own with Electrum by following one of the guides above.
Using a random server is a privacy leak and can be a security issue in some situations. 
Don't Trust, Verify.

closed 2019-09-09 by rootzoll