raspiblitz/raspiblitz #753: Unable to boot fresh installation. RF Kill Switch Status

michaelWuensch, 2019-09-06:
Hi,
I downloaded the v.1.3 image and used etcher to write to my SD card.

I tested it on both my new PI4 and my old PI3B+ and both are unable to boot.

Please note that I use standard Raspis, without LCD

The startup log hangs on:
Started Load/Save RF Kill Switch Status.
Any idea what might cause this? (I also tried it last week with v.1.3-RC5 on my raspi 4 with the same error)
[image]

fluidvoice, 2019-09-06:
It's normal. Without an LCD you won't see what's going on unless you SSH into your node.
Have you tried?

michaelWuensch, 2019-09-06:
No, I haven't tried that yet. I will try it when I am home again.
I don't have a built in LCD, but I connected an external monitor through hdmi, shouldn't that monitor display the same as the LCD?

rootzoll, 2019-09-09:
The RaspberryPi can either output the video on the LCD or the HDMI port - not both. I will try to add a config script for the next release to make the switching between both more easy #767

Aki113s, 2019-09-15:
I have the EXACT same issue, I have the built in LCD screen. That screen shot helped me figure out what was going on.

slc1975, 2019-11-13:
I’m having this issue.. and when I try to ssh with the password it say sorry try again ?

rootzoll, 2019-11-15:
@slc1975 whats the exact command you try to login with ssh? You need to have the user "admin@" before the IP and then password "raspiblitz" should work. Without adding the admin part it tries to login with your local laptop username which is not configured on the RaspiBlitz.

slc1975, 2019-11-28:
ok .. all up and running now.. many thanks .. I would like to get RTL up and running .. but the guild is not very detailed .. you don't fancy doing a howto.. or could you point me in the right direction ? Ta

slc1975, 2019-11-28:
to clarify.. is RTL running on the node or my PC ?

rootzoll, 2019-12-09:
Installing RTL in the main menu go under SERVICES and then set the star under RTL with SPACE and then install with ENTER.

closed 2019-12-09 by rootzoll

dirtysteveo, 2020-01-25:
was this ever resolved? i'm having the exact same issue and i've search every forum and telegram i could find. here are my specs i would love some help. 
Enokay R-Pi 4B Power Supply 5V 3A USB C Charger Adapter
SanDisk Ultra 32GB microSDHC
Crucial MX500 1TB 3D NAND SATA 2.5 Inch Internal SSD
ORICO External Hard Drive Enclosure SATA III
Raspberry Pi 4 Model B 2019 (4GB)
Miuzei Raspberry Pi 4 Touch Screen with Case & Fan

michaelWuensch, 2020-01-25:
@dirtysteveo 
I don't know. At least I didn't get it solved.

rootzoll, 2020-01-29:
With v1.4 there will be a switch to start without LCD - please try that once v1,4 is out and report back. Thanks. https://github.com/rootzoll/raspiblitz/issues/767

SteakDestroyer, 2020-02-14:
I'm having the same issue.
After some poking around, it seems as though when running a Pi4, paticularly with 7" touch screen, the screen never gets past "Load/Save RF Kill Switch Status."
Despite the screen hanging here, the Pi continues to function as normal. Ie. You can SSH to it and do a complete install and run it without a LCD.

Anyone else confirm this issue, and/or isolate what could be causing the problem?

SteakDestroyer, 2020-02-14:
> was this ever resolved? i'm having the exact same issue and i've search every forum and telegram i could find. here are my specs i would love some help.
> Enokay R-Pi 4B Power Supply 5V 3A USB C [...]

Did you ever get it working properly?

dirtysteveo, 2020-02-14:
I never did, I just returned the screen and bought the one on the list and
it works fine. Do you think the new version would support different sized
screens?

mikeblakeuk, 2020-09-04:
I needed the LCD display attached.

mroxso, 2021-07-22:
i have the exact same issue. i have no LCD laying around here so it's only a Pi4..

john-clark, 2024-01-28:
2023 same issue. Not using raspblitz, but official os.