raspiblitz/raspiblitz #756: Node is Syncing (212) 99.99 % for 16+ hours after update to raspiblitz 1.3

goodc0re, 2019-09-06:
After the update to 1.3, both bitcoind as well as lnd are running but the LCD showed 99.9% sync status for bitcoind and lnd, still showed the same status after waiting 16 hours and more and after trying a manual reboot. 

It is not possible to reach the RTL web interface. 

When checking via ssh, the command raspiblitz does not start the info screen: 

$ raspiblitz
Starting the main menu ...
checking bitcoind - please wait ..
can take longer if device was off or first time

it remains like that unless I ctrl-c

Logs: 

https://termbin.com/7g4x
https://termbin.com/coug

I manually stopped lnd and bitcoin and rebooted: 

After entering the lnd wallet passwd, the LCD shows: 
The bitcoind service is not running.   │                                                                                             
│  Login for more details:                │                                                                                             
│  Use CTRL+c to EXIT to Terminal

New logs after the reboot: 
https://termbin.com/mquu
https://termbin.com/rcdc

RTL is redirecting to http://192.168.0.113:3000/rtl/unlocklnd
Unlock/Initialize Wallet

$ raspiblitz
Starting the main menu ...
checking bitcoind - please wait ..
can take longer if device was off or first time

LCD status not starting for a long time, then back to:

Node is Syncing (212) ──────────┐                                                                                             
│  Blockchain Progress :  99.99 %         │                                                                                             
 │  Lightning Progress  :  99.99 %         │                                                                                             
 │  Please wait - this can take some time  │                                                                                             
 │  Use CTRL+c to EXIT to Terminal

fluidvoice, 2019-09-06:
Probably Tor related. The logs for Tor are in your syslog...

`sudo grep -i tor /var/log/syslog | nc termbin.com 9999`

goodc0re, 2019-09-06:
Yes, I see some problems there: 

[warn] /run/tor is not owned by this user (bitcoin, 1002) but by debian-tor (111). Perhaps you are running Tor as the wrong user?

[warn] Before Tor can create a control socket in "/run/tor/control", the directory "/run/tor" needs to exist, and to be accessible only by the user and group account that is running Tor.  (On some Unix systems, anybody who can list a socket can connect to it, so Tor is being careful.)

goodc0re, 2019-09-07:
RTL works now and I can see that all channels are inactive.

rootzoll, 2019-09-09:
Does `nyx` shows active connectivity to other peer nodes?

goodc0re, 2019-09-16:
Lightning Progress  :  99.73 %         

That's less progress than 3 days ago.

rootzoll, 2019-09-17:
Then it looks like the LND somehow stoped syncing and is falling behind relative to growing blockchain. Have you tried to turn it off and on again?

goodc0re, 2019-09-18:
99,69 % today, yes it was still falling behind. 

A reboot helped!

closed 2019-09-18 by goodc0re

Smiggel, 2020-05-24:
@goodc0re Have you fixed this issue? I seem to have the same issue at the moment. It ran fine and suddenly it's lacking...

diogopms, 2020-09-28:
I'm having this issue too, I'm using raspiblitz 1.16