raspiblitz/raspiblitz #758: Waiting to close channel with no on-chain funding TXID

victorscones, 2019-09-06:
I originally tried to open a channel several months ago, but it was stuck on pending to open for nearly a week, so I tried to close it (via the ZAP). This channel has now been stuck on waiting to close since May.  There is no on-chain record of a funding TXID associated with this channel. The block explorer simply shows 'Transaction not found'.  The funds for this channel are still shown under pending balance on ZAP as well.

I would like to upgrade to 1.3 using TOR, so I closed out all my channels and there's one channel left that I'm unable to close. How would I close this channel?

rootzoll, 2019-09-09:
Not sure if this is a ZAP or an LND thing. Can you see that channel still open also when you call the `lnchannels` in the RASPIBLITZ main menu?

victorscones, 2019-09-10:
I don't see the channel listed in `lnchannels`. ZAP, RTL, and `lncli pendingchannels` show the channel as waiting to close. I believe I opened the channel using RTL with a fee of 1 sats/byte. After a few days of waiting for the channel to open, I noticed it was still pending open, so I tried force closing the channel using ZAP. Since then the channel has been waiting to close. `lncli pendingchannels` shows a channelpoint, but there is no record of a TXID when I check a block explorer.

rootzoll, 2019-09-10:
Hmmm .. If your channel was still waiting to open (getting the open transaction into the blockchain) I wonder how it was even possible to close it (never had that situation). I think you should head over to the LND Dev Slack and ask how this edge case is normally handled and see what options are possible in that case: https://dev.lightning.community/

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll