raspiblitz/raspiblitz #760: Feature request - BTC RPC explorer

Tracachang, 2019-09-06:
I think it will be very useful to add BTC rpc explorer,  what do you think? Creating a hidden service it allows to connect to your private explorer through tor.

https://github.com/janoside/btc-rpc-explorer

rootzoll, 2019-09-09:
Sounds like a nice to have but is not on my personal prio list ... but if someone builds a `/home/admin/config.scripts` style prototype script to in- and deinstall such a service I could accept/add it as a PR.

openoms, 2019-10-27:
Just tested this, easy to setup. 
Only it will need `txindex=1` (indexing took ~7h on a RPi4 4GB + SSD)
Setup steps:

`sudo nano /mnt/hdd/bitcoin/bitcoin.conf`  
change txindex to:
```
txindex=1
```

`sudo systemctl restart bitcoind`  

Install:
```bash
npm install -g btc-rpc-explorer
/usr/local/lib/nodejs/node-v10.16.0-linux-armv7l/bin/btc-rpc-explorer
export PATH=$PATH:/usr/local/lib/nodejs/node-v10.16.0-linux-armv7l/bin/
sudo ufw allow 3002
```
Run:
`btc-rpc-explorer -i 0.0.0.0 -u raspibolt -w <passwordB>`

```
  btcexp:app Starting BTC RPC Explorer, v1.1.2 +0ms
  btcexp:app Connecting to RPC node at 127.0.0.1:8332 +4ms
  btcexp:app Connected via RPC to node. Basic info: version=180100, subversion=/Satoshi:0.18.1/, protocolversion=70015, services=000000000000040d +284ms
```

Can be viewed in a browser on:
RaspiBlitz_LAN_IP:3002
[image]

openoms, 2019-11-04:
To add to the PATH permanently:

`sudo bash -c "echo 'PATH=\$PATH:/usr/local/lib/nodejs/node-v10.16.0-linux-armv7l/bin/' >> /etc/profile"`

The setup for [LND-admin](https://github.com/janoside/lnd-admin) is basicly the same:
```
npm install -g lnd-admin
lnd-admin
Open http://127.0.0.1:3004/
```

openoms, 2019-11-08:
Made a scripted version: https://github.com/openoms/raspiblitz-extras#btc-rpc-explorer
but could not make the `.env` file work yet.
From the README:

>Configuration
Configuration options may be passed as environment variables or by creating an env file at ~/.config/btc-rpc-explorer.env or at .env in the working directory. See .env-sample for a list of the options and details for formatting .env.

Running as the `bitcoin` user and have tried    
`/home/bitcoin/.config/btc-rpc-explorer.env`   
and  
`/usr/local/lib/nodejs/node-v10.16.0-linux-armv7l/lib/node_modules/btc-rpc-explorer/.env`  
where the .env-sample is, but there is no effect.

Any ideas why my .env file is not working?

openoms, 2019-11-18:
PR is in: https://github.com/rootzoll/raspiblitz/pull/867

openoms, 2019-11-24:
Merged now after successful tests https://github.com/rootzoll/raspiblitz/commit/bf480bd721fb7079cc8a05aefd99c9a85d09b406

closed 2019-11-24 by openoms