raspiblitz/raspiblitz #761: What is the minimum set of files to reuse the Bitcoin blockchain?

jayrmotta, 2019-09-07:
Hey guys,

I have an external HD that I've downloaded the Bitcoin blockchain in the past, I'm trying to restore my node to recover some lost funds but I'm facing some issues that seem to be related to the files in the HD.

Recently I've burned the new version of Raspiblitz into a micro SD card, but given the state persisted in the HD when it started the node was "recovered", but I wanted to go through the process of recovering (because I've lost some on-chain funds in an old node), so I mounted the HD and decided to move some files away to avoid having to download and sync the Bitcoin blockchain but at the same time avoid this auto recovery process.

Now the problem is that I keep having issues like this `Unable to create chain control: unable to subscribe for zmq block events: dial tcp :28332: connect: connection refused` message or many others I've seen and disappeared after rebooting the node many times.

Sorry for the long intro, but I felt I needed to give a proper context. I want to know what is the minimum set of folders and files I should keep in the external HD, so I can start a node as if it never existed.

rootzoll, 2019-09-09:
Normally if you just want to use the blockchain data of an HDD to setup a node you can (let it to its "recovery" and then) use the "REPAIR" menu "RESET-HDD". See https://github.com/rootzoll/raspiblitz#repair-options-to-test-repair-and-reset-your-raspiblitz

Now that you delete stuff manually from the HDD I am not sure whats maybe missing. The minimum folders should be: `bitcoin/blocks` & ` bitcoin/chainstate`

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll