raspiblitz/raspiblitz #764: [Feature Idea] Remote access to Raspberry Pi with serveo.net

cryptomulde, 2019-09-09:
A few days ago I stumbled upon a service called serveo.net

It allows you to remotely connect to ports without having to expose your local network ports.
I wrote a guide on medium.com
https://medium.com/@cryptomulde/how-to-use-serveo-net-to-connect-to-your-local-raspiblitz-server-cd6d61834978

Maybe at one point we can make this a service feature in the Raspiblitz? What do you guys think?

rootzoll, 2019-09-09:
>"Serveo is an SSH server just for remote port forwarding. When a user connects to Serveo, they get a public URL that anybody can use to connect to their localhost server."

The RaspiBlitz already supports SSH Reverse-Portforwarding ... see:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-setup-port-forwarding-with-a-ssh-tunnel

So basically with OpenSSH (that is installed on almost every Virtual Server you can rent) you can alrready do port-forwarding without any extra software.

serveo is not open source and just offers a "free" version that is limited to 3 simultan connections ... so not everybody is free to host services.

cryptomulde, 2019-09-09:
Cool! Thanks for clarification. We can probably close this :)

closed 2019-10-03 by frennkie