raspiblitz/raspiblitz #765: Failing RaspiBlitz - trying to download back up file - noob

hch-y, 2019-09-09:
I've had issues with my node for a couple months now, probably due to user error and maybe a failing HHD(had this issue at first https://github.com/rootzoll/raspiblitz/issues/442). Now I've decided to download the channel back up file and start fresh.  Due to serious stupidity on my part I may have committed some fatal errors and have no shot at recovering my funds. I'll do my best to describe where I'm at, any help would be greatly appreciated.

1) Went to ssh into my node and realized that the last time I had tried to fix my node by reflashing my SD card my IP for the node had changed. Of course I was too lazy to find where I'd written it down so I just plugged in a keyboard directly to my pi and and ran /home/admin/config.scripts/lnd.rescue.sh backup. This seems to have worked and stopped my node. 

2) And now realize that I really screwed up because I need to download the file somewhere. So I dig around and find my IP for the node and try to sign in the proper way through terminal on my laptop. This just times out. I'm guessing this is because I've stopped my node and can't access it. Not sure where to go from here.

3) I have tried plugging in a keyboard straight to the pi along with a thumb drive and running scp -r admin@home/admin/lnd-rescue-*.tar.gz ./ but it states that can't resolve the hostname(where it wants to download file?). How do I resolve the hostname issue? I tried running scp -r admin@[NAME OF THUMBDRIVE]/admin/lnd-rescue-*.tar.gz ./ with no success. 

To anyone who read along, thank you very much and any help would be greatly appreciated. At the end of the day I'd really just like to have a fresh node with my funds. Yes, I had funds in open channels when this began and I do have my seed phrase.

rootzoll, 2019-09-09:
@home is not a valid address to download something from. But lets start at the beginning - I assume you have v1.2 running.

You want the channel backup file ... so lets see if its still available. Please check if under the following locations is still a file called: `channel.backup`

`sudo ls /home/admin/.lnd/data/chain/bitcoin/mainnet/` or `sudo ls /mnt/hdd/lnd/data/chain/bitcoin/mainnet/`

Is there still `channel.backup` file in any of these directories?

hch-y, 2019-09-09:
@rootzoll thanks for the response! 

I just ran both of those and I get a message saying that there is "no such file or directory" for each one. My external hard drive might be completely fried or something.

rootzoll, 2019-09-09:
The first path `home/admin/.lnd/data/chain/bitcoin/mainnet/` is just the sd card ... if that is gone it looks not that promising to get to a restore.

hch-y, 2019-09-09:
hmm, that's bad. anything I can do at this point?

hch-y, 2019-09-10:
Made some progress, screen shot attached of bug report,
[image]

rootzoll, 2019-09-10:
Can you post a screenshot of the following directory listing?
```
sudo ls /mnt/hdd
```

hch-y, 2019-09-10:
[image]

rootzoll, 2019-09-10:
Thats not too bad to start with. Whats inside the lnd folder and the lnd backup folder?
```
sudo ls /mnt/hdd/lnd
sudo ls /mnt/hdd/backup_lnd
```

hch-y, 2019-09-10:
[image]

hch-y, 2019-09-10:
oops, ran sudo ls /mnt/hdd/backup_lnd correctly below  

[image]

rootzoll, 2019-09-10:
I am seeing a data dir :) ... can you try also:
```
sudo ls /mnt/hdd/backup_lnd/data/chain/bitcoin/mainnet/
```
and if that is not working ... remove step by step the ending directoty until its working?

hch-y, 2019-09-11:
[image]
[image]

hch-y, 2019-09-11:
[image]

rootzoll, 2019-09-17:
OK so the backup_lnd seems from an old update that still dont contains the channel backup file.
You were running RaspiBlitz v1.2 now correct?

Lets give it one more try to see whats in the original LND folder:

```
sudo ls /mnt/hdd/lnd/data/chain/bitcoin/mainnet/
```

and if that is not working ... remove step by step the ending directoty until its working?

hch-y, 2019-09-26:
Yes, I am running v1.2.  

[image]
[image]

and then, if run the commad below it appears there is some sort of backup file.
[image]

I just don't know how to actually download it to my laptop.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll