raspiblitz/raspiblitz #766: Renew TLS certs on IP change

rootzoll, 2019-09-09:
see comment by @openoms on #324 :
https://github.com/rootzoll/raspiblitz/issues/324#issuecomment-527331964

Users have been reporting that the they could not connect Zap after chnaging their IP.
Manually resetting the TLS.cert helped.

Should this be automated every time when the publicIP is triggered to changed in lnd.conf
by the

```
 # RECHECK PUBLIC IP
 # when public IP changes, restart LND with new IP
```

function of _background.sh?

rootzoll, 2020-03-16:
Movin to v1.6 because just an automatic renew of TLS without notifying the user is also confusing and not a good fix. Maybe when LCD interface is more advanced this can be a "notification".

rootzoll, 2020-05-27:
Moving to v1.7 because 1.6 will not have any new LCD-interface features

NilByte, 2020-12-24:
When you run duck.dns.org, the scan code for Zap is generated with your public ip address but  Zap wallet will only load code edited to use your domain name instead of public ip.

openoms, 2024-01-22:
There is a request to renew the TLS cert on expiry for Electrs also: https://t.me/raspiblitz/140628

This is not covered with the reset script if LND is not installed.

rootzoll, 2024-03-19:
Tests good on my machine. Closing issue for release.

closed 2024-03-19 by rootzoll