raspiblitz/raspiblitz #768: Blockchain downloaded by blockchain lost

zionsuperfly, 2019-09-10:
I downloaded the blockchain by torrent. Both (lnd and blockchain) where finished with downloading and hashing. I opened the session with my phone and it started to delete and move block. But my session was not staying open (cell phone went in standby). Now the system will try to redownload again. How can i recover the downloaded blockchain?

patrikmojzis, 2020-03-22:
I had the same problem. What did you do then? Did you re-download it again or synced up it via BTC network?

rootzoll, 2020-03-23:
Terminals closing/standbying can be problem, thats why using a terminal on dekstop is recommended for now.

Once we will have the TORBrowser-Setup such processes will be designed to run in the background.