raspiblitz/raspiblitz #773: How do I start from scratch?

charlierlee, 2019-09-11:
If I remove the partitions to start from scratch, and start from a fresh sd card I get this:

Starting the main menu ...
***********************************************************
WARNING: HDD HAS NO PARTITIONS
***********************************************************
Press ENTER to create a Partition - or CTRL+C to abort

Creating Partition ...
Warning: The resulting partition is not properly aligned for best performance.
...
***********************************************************
WARNING: bootstrap still updating - close SSH, login later
To monitor progress --> tail -n1000 -f raspiblitz.log
***********************************************************

Then it reboots and says:
Give root password for maintenance or press control-D to continue.

This happens every time.

openoms, 2019-09-11:
If you just delete partition when it is recreated all the old data appaears on the disk. Delete the partition, recreate and format the disk to NTFS for example. This way the RaspiBlitz setup will format it during the setup and you will have an empty disk.

charlierlee, 2019-09-12:
Thanks for your help. That fixed it!

closed 2019-09-13 by rootzoll