raspiblitz/raspiblitz #774: [v1.3] The bitcoind service is not running.

zionsuperfly, 2019-09-11:
[image]

zionsuperfly, 2019-09-11:
https://termbin.com/7k6p

Tracachang, 2019-09-13:
This Rolling forward "issue" can happen when bitcoind has not been stopped properly, you just have to wait, it takes long time but when finish thia process bitcoind will start.

belgacoin, 2019-09-20:
Hello, I got the same error after upgrading from 1.0 to 1.3. So I just leave my raspi on while on that screen?

Tracachang, 2019-09-20:
Yes, follow the progress with:

sudo tail -f /mnt/hdd/bitcoin/debug.log

belgacoin, 2019-09-20:
That command gives me "no such file or directory" tail: "no files remaining" :-o

Tracachang, 2019-09-20:
You don't have a debug.log in /mnt/hdd/bitcoin?

belgacoin, 2019-09-21:
I guess not. This is what I get:
~ $ sudo tail -f /mnt/hdd/bitcoin/debug.log
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
tail: no files remaining

Tracachang, 2019-09-21:
Then bitcoind is not running,  when starts it creates the debug.log file. Start bitcoind manually.

belgacoin, 2019-09-21:
I started bitcoind manually, next screen is:

****************************************
* The LND service is not running.
****************************************
If you just started some config/setup, this might be OK.

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

$ sudo tail -f /mnt/hdd/bitcoin/debug.log gives me:

2019-09-21T11:29:41Z UpdateTip: new best=0000000000000000000e6042507456e17a84b838bafbcd84b1d9ca0de9b719f6 height=595744 version=0x20c00000 log2_work=91.1106 tx=457030967 date='2019-09-20T09:13:34Z' progress=0.999238 cache=6.1MiB(53308txo)
2019-09-21T11:29:41Z New outbound peer connected: version: 70015, blocks=595913, peer=20
2019-09-21T11:29:59Z UpdateTip: new best=00000000000000000014e41887a6d0e68d0ce3d20eb7b48407ab60e82b4f7093 height=595745 version=0x20000000 log2_work=91.110628 tx=457032645 date='2019-09-20T09:14:03Z' progress=0.999238 cache=6.4MiB(56052txo)
2019-09-21T11:30:08Z UpdateTip: new best=0000000000000000000b358342ae22251b78179072ffe9982a485541b54e281a height=595746 version=0x3fffe000 log2_work=91.110656 tx=457033529 date='2019-09-20T09:16:21Z' progress=0.999239 cache=6.6MiB(57860txo)
2019-09-21T11:30:08Z Pre-allocating up to position 0x6000000 in blk01800.dat
2019-09-21T11:30:34Z Pre-allocating up to position 0xa00000 in rev01800.dat
2019-09-21T11:30:34Z UpdateTip: new best=00000000000000000013a0adad5eaae6453dee705931300df6a472e8661ff438 height=595747 version=0x20000000 log2_work=91.110683 tx=457035421 date='2019-09-20T09:22:16Z' progress=0.999242 cache=7.0MiB(61851txo)
2019-09-21T11:30:34Z New outbound peer connected: version: 70015, blocks=595913, peer=27
2019-09-21T11:30:50Z UpdateTip: new best=0000000000000000000f717e8839baa6aa7461e653404ea5d0364108f4a14a65 height=595748 version=0x365ba000 log2_work=91.110711 tx=457036464 date='2019-09-20T09:24:59Z' progress=0.999243 cache=7.3MiB(63997txo)
2019-09-21T11:30:50Z New outbound peer connected: version: 70015, blocks=595913, peer=30

belgacoin, 2019-09-21:
XXdebuglogs: https://termbin.com/yxcs

Tracachang, 2019-09-21:
I see bitcoind is running and synched, just unlock lnd wallet and all should be fine.

In terminal : lncli unlock

belgacoin, 2019-09-22:
hm, again I have an error:
lncli unlock
Input wallet password: 
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

MaxexaM, 2019-09-22:
I had the same problem (bitcoind not running) I solved by changing the SD with a new one.

MaxexaM, 2019-09-22:
I have the same problem, re-installed many times, replaced SD card ... everything works, the blockchain starts to download but then suddenly it stops at 8% or 10% ... could it be the SSD?

Tracachang, 2019-09-22:
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

Did you manually modified lnd.conf?

belgacoin, 2019-09-23:
No, I just burned the new image.

Tracachang, 2019-09-23:
I have the same error :

[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

I've deleted lnd (in mainmenu -- repair) and lnd doesn't starts properly getting all the time same error.

openoms, 2019-09-23:
> I have the same problem, re-installed many times, replaced SD card ... everything works, the blockchain starts to download but then suddenly it stops at 8% or 10% ... could it be the SSD?

I had similar problems when using an incompatible SATA-USB adapter with my SSD on the RPi4. If you are using the RPi4 make sure you use one which has been tested to work before: https://github.com/rootzoll/raspiblitz/issues/691#issuecomment-520526004

Tracachang, 2019-09-23:
Finally in repair I've deleted hdd (keeping blockchain) installed a fresh flashed raspiblitz 1.3 and all is working.

belgacoin, 2019-09-24:
I'm still offline :( I don't have too much time nor the knowledge to give it a forensic look.

MaxexaM, 2019-10-03:
hi @rootzoll, ok, I tried another HDD, 1 week to download the 100% blockchain and then again "bitcoind not running"
https://termbin.com/kvl7

openoms, 2019-10-03:
@MaxexaM Hi, just looked into you logs. Bitcoind did not finish indexing yet, but almost reached the tip of the blockchain. Will need to give it some time and check again in the logs.

When you see this in the system status:
```
bitcoinErrorShort='Loading block index' bitcoinErrorFull='error code: -28 error message: Loading block index...'
```
means it is working. Might be only a couple of hours.

MaxexaM, 2019-10-03:
@openoms ok ! thanks... I'm wait it!!

openoms, 2019-10-03:
The bitcoind not running error message is inaccurate. `Bitcoind is indexing` would be more correct according to your logs.

MaxexaM, 2019-10-04:
@openoms yeahhh !! it works! :) thanks for your patience ... is there a tutorial to know how to use the node well?

openoms, 2019-10-05:
> @openoms yeahhh !! it works! :) thanks for your patience ... is there a tutorial to know how to use the node well?

Glad it works! Apart from the readme it is good to skim through the FAQ: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md and there is this repo for some introduction to using the LN node: https://github.com/openoms/lightning-node-management/blob/master/README.md

belgacoin, 2019-10-08:
Hello, my node is still offline. 

sudo systemctl start bitcoind.service
sudo tail -f /mnt/hdd/bitcoin/debug.log
2019-10-08T16:17:37Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-10-08T16:17:37Z Using 4 threads for script verification
2019-10-08T16:17:37Z scheduler thread start
2019-10-08T16:17:37Z Wallet disabled!
2019-10-08T16:17:37Z Binding RPC on address 127.0.0.1 port 8332 failed.

I also have another window open that says:
2019-10-08T16:18:31Z UpdateTip: new best=000000000000000001630b64499eec679dafbb73893bfdbc0447c5ac14c70367 height=368083 version=0x00000003 log2_work=83.158864 tx=78497488 date='2015-08-02T10:27:30Z' progress=0.171356 cache=644.5MiB(5760517txo)
2019-10-08T16:18:37Z ThreadRPCServer incorrect password attempt from 127.0.0.1:55320

2019-10-08T16:17:37Z Unable to bind any endpoint for RPC server
2019-10-08T16:17:37Z Error: Unable to start HTTP server. See debug log for details.
2019-10-08T16:17:37Z Shutdown: In progress...
2019-10-08T16:17:37Z scheduler thread interrupt
2019-10-08T16:17:37Z Shutdown: done

openoms, 2019-10-09:
@belgacoin 
compare your bitcoin conf:
`sudo nano /mnt/hdd/bitcoin/bitcoin.conf`
with the default: https://github.com/rootzoll/raspiblitz/blob/master/home.admin/assets/bitcoin.conf
There might be some manual changes causing you issue.

Fallback option is to backup your lnd directory with:
`/home/admin/config.scripts/lnd.rescue.sh backup`
and start with a clean SD and HDD too.
To clean the HDD you need to format it on an machine, NTFS filesystem on windows would do.
You will be able to restore your lnd from the backup file with the help of the initial setup.

belgacoin, 2019-10-09:
@openoms There were indeed some changes I had to make. Now I have restarted and I'm waiting  "bitcoind is starting" has finished (it does mention "error found in logs" too).
Now caching up some 1000 blocks. We'll see after that.

Question: if I have to unlock lncli again, is there a default password? Nothing worked.
Thanks!

openoms, 2019-10-09:
@belgacoin there is no default password for the LND wallet only the one you give (twice) when creating the wallet. Try unlocking again only when bitcoind is synced.

belgacoin, 2019-10-11:
Hello @openoms, I notice that the update process gets stuck all the time:
2019-10-11T13:12:46Z UpdateTip: new best=000000000000000000046942c8acddaf7d6ce230850d389e7bbbcc0b8e6be413 height=598918 version=0x20000000 log2_work=91.199766 tx=463900896 date='2019-10-11T13:12:17Z' progress=1.000000 cache=138.2MiB(1226619txo) warning='46 of last 100 blocks have unexpected version'
2019-10-11T13:24:54Z socket recv error Connection reset by peer (104)

When going to the main screen with "raspibliz" command, it tell me that LND service is not running. When do "lncli unlock" the input password never matches:

lncli unlock
Input wallet password: 
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

XXdebuglogs are here: https://termbin.com/4cls

openoms, 2019-10-11:
@belgacoin Your bitcoind is synced and looks ok.
the Lnd service is disabled:
```
/etc/systemd/system/lnd.service; disabled; 
```
run:
`sudo systemctl enable lnd`
`sudo systemctl start lnd`

check status with:
`sudo systemctl status lnd`

or with the XXdebugLogs.sh

belgacoin, 2019-10-11:
Thanks @openoms , that seems to work. https://termbin.com/movy
LND meanwhile successfully unlocked and is syncing atm :)

closed 2019-11-20 by rootzoll

chuta, 2020-04-02:
> The bitcoind not running error message is inaccurate. `Bitcoind is indexing` would be more correct according to your logs.

i fully agree with you.