raspiblitz/raspiblitz #776: tor bitcoin.conf.lnd.conf settings improvements

Tracachang, 2019-09-13:
Hello,

Setting up raspiblitz over tor the and checking bitcoin.conf and lnd.conf I do believe some important arguments are missing.

The actual bitcoin.conf over tor setup adds:
- nodes
- onlynet=onion 

I would add:
proxy=127.0.0.1:9050 
bind=127.0.0.1 
dnsseed=0 
dns=0 

Deactivating DNS would avoid Sybil attacks and since you already added nodes, there is no need activate it to find some peers.

For lnd.conf and based on https://github.com/lightningnetwork/lnd/blob/master/docs/configuring_tor.md I would add to lnd.conf:

tor.active=1 
tor.v3=1
tor.streamisolation=1 
listen=localhost

I've also seen that bitcoin user is not in debian-tor group and all guides I've read so far advise to do it, however didn't have any error. So I am not sure what is the benefit.

Thank you.

Tracachang, 2019-09-13:
I have been doing tests and all works fine.

I have edited /mnt/hdd/bitcoin/bitcoin.conf and my tor settings are (the orginal options are all the nodes and onlynet=onion:

proxy=127.0.0.1:9050
bind=127.0.0.1
onlynet=onion
dnsseed=0
dns=0
(and all the nodes)
addnode=fno4aakpl6sg6y47.onion 

I have edited lnd services
sudo nano /etc/systemd/system/lnd.service

ExecStart=/usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735

(as you can see I have changed:
 --tor.v2 for --tor.v3
added --tor.streamisolation

Tracachang, 2019-09-13:
I've edited the tor script and sent a PR.

openoms, 2019-09-15:
Related to the discussion in: https://github.com/rootzoll/raspiblitz/issues/237

rootzoll, 2019-09-17:
Thanks for the PR ... will ne part of issue #237 ... closing this now.

closed 2019-09-17 by rootzoll