raspiblitz/raspiblitz #780: Rasp Pi 4 Tempretures

Aki113s, 2019-09-15:
I have an LCD screen on my Pi case, on initial boot for Raspiblitz I got a warning about temps, I then read about Pi 4 having temp issues and there is a firmware fix, but some say it has USB issues, basically....do I need to worry about temps here??? I have aluminum heat spreaders on the chips. It blinks, I guess the Pis just run hotter than normal?? I really wanted to use an LCD to view without needing to SSH into device to check on things.

The initial system boot gave me about 72-75 degrees, which I don't think is bad. 4GB Raspberry Pi 4 model.

Tracachang, 2019-09-15:
Running high temperatures for prolonged periods of time can not be good. On the best case you'll have lower performance and in the worst it can damage your raspberry pi.

Did you think buying a fan shim?  I got one and it is very efficient,  I've setup between 55-65°C

Aki113s, 2019-09-15:
What I was figuring, I was hoping to run an small LCD screen on the Pi but I don’t see that happening if I need a fan. Which I do have. One the blockchain syncs I think I’ll swap cases and then SSH in 

:(

Tracachang, 2019-09-15:
Why not? You can use LCD with fan shim

Aki113s, 2019-09-17:
> 
> 
> Why not? You can use LCD with fan shim

I ordered one, Got a USB fan going and dropped it from 83 to 55 until it arrives. Thanks for that insight!

closed 2019-09-17 by rootzoll