raspiblitz/raspiblitz #781: Ok to stop initial sync?

Aki113s, 2019-09-15:
With the LCD screen on I cannot use a fan and the aluminum heat syncs seem to do little. 

I just got a cheap case and small fan that should keep me at speeds that won’t throttle. The Blockchain is syncing, half way to go. I imagine all cores are engaged on the initial sync in overdrive so the extra temps could be from this but since the LCD is not working properly, stays on boot screen, I may as well just go to a case and fan. 

Can I just power off the Raspiblitz? Swap cases turn it back on and the sync continue?? I elected to sync and not FTP, I’m brand new to this so I don’t want to do anything wrong. Should I let the Pi run at 83 degrees until the Sync is done then do the swap??

Thank you!

Tracachang, 2019-09-15:
Before power off you must be sure that lnd & bitcoind are not running to avoid blockchain corruption, to do it write in terminal:

sudo systemctl stop lnd
sudo systenctl stop bitcoind

rootzoll, 2019-09-17:
> With the LCD screen on I cannot use a fan and the aluminum heat syncs seem to do little.

Check the ShimFan from the new v1.3 shopping lists ... it will fit between LCD and Raspi. 
Closing this issue.

closed 2019-09-17 by rootzoll