raspiblitz/raspiblitz #782: Give root password for maintenance (or press Control-D to continuine): = Message after new Installation Vers. 1.3 - SSH is not working

Otti24xx, 2019-09-16:
after new Installation Vers. 1.3 i get this message on display - SSH is not working, what can i do?

joernroeder, 2019-09-16:
port 22 is closed after i've run the cleanup script `./XXcleanHDD.sh` as well.
The display says i need to login via ssh but `nmap` returns `22/tcp    closed ssh` for the given ip.

rootzoll, 2019-09-17:
If you get this "maintenance" notice something got corrupted on the sd card. Best solution is to shutdown (cut power if needed) and write a fresh image to the sd card. On next start you will go thru update/recover process and have to set the Password A again ... but other then that your RaspiBlitz will use your HDD data and continue with your old configuration and channels.

joernroeder, 2019-09-18:
@rootzoll i did not came across the "maintenance" notice, and had to refresh the sd card a couple of times before it finally worked and i was able to restore my old state. the strange thing is that port for bitcoin ´8333` as well as the webinterface `3000` were still open while `22` was `closed` or `filtered`.

Otti24xx, 2019-09-19:
Possibly it was a HDD Error, I have now a new SSD, so it works fine!

raziel8, 2019-10-29:
I'm stuck in this step. After like 10 times with different sd cards (just to be sure) and many tries, no way of rebooting without this problem.

Any suggestion?

raziel8, 2019-10-29:
@rootzoll the update/recovery process seems to works but in the end, it still fails after the reboot

rootzoll, 2019-11-16:
If you disconnect the HDD/SSD on boot up - does it also give that that error, or another?

flowthis, 2019-11-20:
I`ve been running into the same problem. 

Disconnecting the HDD on bootup results in a brief period in which the following
message is repeated. 

"A start job is running for /dev/dis 96bf..."

It ends after approx. 60 seconds with the initial problem prompt. "Give root passw..."

Have you been able to find a solution @raziel8 ? 
Are there any possible workarounds that may let me reclaim my, otherwise very reliable node @rootzoll ?

Otti24xx, 2019-11-20:
Hi, i have changed my hdd to SSD – so now it works fine, I think it was a Problem with my HDD.

Best Regards
Johohannes

Von: flowthis
Gesendet: Mittwoch, 20. November 2019 12:03
An: rootzoll/raspiblitz
Cc: Otti24xx; Author
Betreff: Re: [rootzoll/raspiblitz] Give root password for maintenance (orpress Control-D to continuine): = Message after new Installation Vers. 1.3 -SSH is not working (#782)

I`ve been running into the same problem.
Disconnecting the HDD on bootup results in a brief period in which the following
message is repeated.
"A start job is running for /dev/dis 96bf..."
It ends after approx. 60 seconds with the initial problem prompt. "Give root passw..."
Have you been able to find a solution @raziel8 ?
Are there any possible workarounds that may let me reclaim my, otherwise very reliable node @rootzoll ?

rootzoll, 2019-11-20:
@flowthis is this also happend on an update or during normal operations/reboot?

codeoholic, 2019-11-26:
@rootzoll This was the last message I received.
DONE - Give raspi some cool off time after hard building .... 20 secs sleep
END Provisioning
rebooting

Post it, I see a prompt for root password for maintenance.

rootzoll, 2019-11-26:
Have you connected a USB keyboard and entered the root password? Should be the same passwordA as for the admin user.

codeoholic, 2019-11-26:
@rootzoll Does not accepts Password A. It takes raspiblitz as the root password and shows terminal prompt in root directory.

rootzoll, 2019-11-26:
Ah yeah - you getting this error directly after a recover/update, thats when PasswordA is not yet reset and still the default "raspiblitz".

Try also the CTRL+D or follow the hints in this RaspberryPi forum:
https://www.raspberrypi.org/forums/viewtopic.php?t=176695

So it maybe that somehow after recover/update your system is not shutting down properly and you have to fix Fstab all the time. If you get it fixed it would be interesting if this is just happening on this recover or every reboot/shutdown.

codeoholic, 2019-11-27:
@rootzoll  CTRL+D does nothing. Keeps prompting for password over and over.
Giving the root password, I am not sure what to be removed from fstab.
One is /boot, second is / and the third is /mnt/hdd

PapaCoderNit, 2019-11-27:
Im Having the same situation.. But im able to login bash with raspiblitz password and able to edit it.. What should my Fstab file be??

codeoholic, 2019-11-27:
@PapaCoderNit Are there any unix commands as well to perform the same actions? I am on MacOS.

PapaCoderNit, 2019-11-27:
> @PapaCoderNit Are there any unix commands as well to perform the same actions? I am on MacOS.

I did it with usb keyboard plugged into the raspberry pi.
Also you want to do your sd card not the hdd..
Check /etc/dev/   and youll see sba and sba1 
The sd card is most likely sba
Run sudo fsck /etc/dev/sba (make sure its your sdcard)

rootzoll, 2019-12-12:
To prevent this scenario in the future I will add the command `sudo tune2fs -c 1 /dev/mmcblk0p2` into the sd card build script. This should force a fsck automatically on the sd card root partition on every boot and hopefully gets rid of this "maintenance login" screen.

raziel8, 2019-12-16:
@flowthis not yet, just came back on this project! Keep you posted

@rootzoll I saw you pushed two commits to solve this issue, let's give a try now. 
Just to be clear: I'm using an HDD, not a SSD

rootzoll, 2019-12-17:
@raziel8 to have the #782 commit active you need a sd card build with the updated script. A release candidate sd card image of v1.4 will be out the next days - that one will have the commit included.

raziel8, 2019-12-17:
@rootzoll thank you so much, waiting for the v1.4 ✌🏼

PeerZetZzZzZ, 2020-03-23:
> fsck

I've stucked on the same same message after simply plug off during BtcPAY server setup which just hanged.
How I resolved:
1. Connect SD card to another Linux
2. Run GParted partition program, selected partitions and check "Check" where possible (on every partition it's possible).
3. Run it (it showed the window about possible data loss but I confirmed).
4. After 10 seconds operation done, SD card back to raspberry - no errors anymore.

Hope it can help, but beware if you are scared to lose data on SD card.

rootzoll, 2020-03-23:
As it looks the v1.4 is doing also now more fsck .. but its still asking for manual interference.
Thats why the v1.5 will try if this autofix-flag will no longer showing you this message.
https://github.com/rootzoll/raspiblitz/issues/1053

edgeball, 2020-03-28:
yes, I've been having the same issue since 1.3. Now Upgraded to 1.4 but still. Is there a way to fix it manually?

joernroeder, 2020-03-29:
@edgeball i fixed it manually by entering the root password (pw a) and appending the proposed fix for v1.5 `fsck.mode=force fsck.repair=yes` manually into `/boot/cmdline.txt`. the reboot afterwards took a while (i think tue to the repair) but it resumed correctly afterwards

rootzoll, 2020-03-30:
#1053 see also

closed 2020-04-18 by rootzoll

josusanmartin, 2020-11-18:
I was running into this issue, and the problem ended up being the SSD case that I was using. I tried putting the SSD in an old case I had and everything went smoothly.

rootzoll, 2020-11-18:
@josusanmartin did you just changed the case and it was running again or did you also made a fresh sd card image. Also can you provide I link to that case?

josusanmartin, 2020-11-18:
This was during the initial sync, so because of the error I ended up creating a fresh image a few times. So I think I created a fresh sd card image when it ended up working too (80% confident since I don't fully remember).

From what I could gather, the "bad drive" was causing the distro to get "corrupted" somehow. Whenever I started completly fresh (formatted ssd and fresh sd image) it would start syncing the blockchain and after a while it would throw io errors. I would reboot and  "the give root password for maintenance" would show up. So I'd create a fresh sd image and it would boot up. But I couldn't get it to start syncing again, so I had to erase the ssd too.

rootzoll, 2020-11-18:
Thanks for the report. I am collecting to see if we can spot a pattern.

raziel8, 2021-01-02:
Thanks for the resolution, I solved changing the USB port, some USB ports can cause this issue also in newer versions apparently

stopfeds, 2021-03-25:
Currently facing the same issue because I removed the power plug after removing the LAN cable - cause it didn't reconnect to LAN.
Will try to flash a new SD card now...

stopfeds, 2021-03-27:
> Currently facing the same issue because I removed the power plug after removing the LAN cable - cause it didn't reconnect to LAN.
> Will try to flash a new SD card now...

Worked for me!

ronledfordjr, 2021-04-21:
Weird, I was having this issue and last night I selected the Auto Unlock Feature for LND, it rebooted and Poof no more CTRL D issue. I rebooted again just now to confirm, no ctrl d issue.

Tibexa, 2021-07-16:
Here you have the solution:
1. Give your password ti login to maint..mode 
2. Just type fsck -f -y / .....to repair your /
3. then reboot
4. And check cmos battery life