raspiblitz/raspiblitz #783: [solved] unable to recover backup due to character limitations

joernroeder, 2019-09-16:
I was unable to restore an old wallet and `channels.backup` due to the restricted character set  – which was introduced after i created the wallet – via the interface.
recovery was in the end successful after bypassing the check and setting `passwordD` directly to the correct value in https://github.com/rootzoll/raspiblitz/blob/5a501e65211c4d62286c551a87742fd304cbc1b7/home.admin/70initLND.sh#L345 

So if someone stumbles over the same issue, this might be a solution – otherwise it might be an idea to allow all characters for restoring purposes.

Thanks for the great work!!!

frennkie, 2019-10-03:
@joernroeder would you mind adding a short section about this to https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz as a `Pull Request`?

nyxnor, 2021-04-18:
@joernroeder Was this solved on later releases? Can you close this?

nyxnor, 2021-05-17:
@rootzoll Please close this.

closed 2021-05-18 by rootzoll