raspiblitz/raspiblitz #784: LCD screen displaying incorrectly (Pi 4)

Aki113s, 2019-09-17:
So my LCD screen on my Pi stays on this screen, ends on the Kill Switch as shown here...its to small to read so this is from a pic I found online but same thing basically.
[image]
I can SSH into the Pi and Blockchain is at 91%, so I am getting there. The next step in the directions says once done, I should see the following.
[image]
I have high doubts based on many reboots my LCD screen sits on the first image, what could be causing it? I assume I should still be able to SSH in to the Pi to verify all is good. I'm so curious as to why it hangs on the boot screen.

Thank you!

rootzoll, 2019-09-17:
We had similar issues in the past ... somehow the LCD gets not updated anymore, but the system is still progressing. We never got exactly behind why that is happening on some screens ... any idea and investigation wellcome. See issue: #195

Aki113s, 2019-09-18:
> 
> 
> We had similar issues in the past ... somehow the LCD gets not updated anymore, but the system is still progressing. We never got exactly behind why that is happening on some screens ... any idea and investigation wellcome. See issue: #195

I posted in #195 as well but also to add here, I read through that thread...
A fix seemed to work for me.

So I was having the EXACT same issue. Fresh first time install. Raspberry Pi 4...4GB, Everything synced. In SSH terminal I ran what @pxsocs said.

**cd /home/admin/LCD-show
sudo ./LCD-hdmi**

System rebooted, saw the entire boot run. Then it went to the loading screen, it stayed on Lightning Preparing Sync and rapidly finished. Then the Display worked, unsure if it will happen again, but will see.

My display is VERY small though, the status screen in 25% of what it should be. Small steps yes?

[image]

Is there a way to full size this?? Again I can SSH in no problem, I just love peaking at the temps and when LND is funded I can monitor easier.

Thanks all!!

Aki113s, 2019-09-18:
I was able to find my answer...
run terminal and type
**sudo raspi-config**
Go to: >Advanced Option > Resolution > just set your resolution compatible fit with your screen.

Apply and let it reboot.

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll

wsurferdude, 2021-09-27:
I have an Official Raspberry Pi 7" Touchscreen and to get it to fill the Info screen I modified the following.
framebuffer_width=500
framebuffer_height=360

Just keep on moving these numbers down to make the display larger.  Don't ask me why this works 😝.