raspiblitz/raspiblitz #785: Optimize Restoring Old wallet from Channel Backup

rootzoll, 2019-09-17:
At the moment the LND channel backup file gets added after the wallet is created. Check if this can be done on wallet crearte ... see:

https://github.com/lightningnetwork/lnd/blob/master/docs/recovery.md#recovering-using-scbs

Also re-check if n-chain & channels funds get recovered ... see fail reports:
https://github.com/rootzoll/raspiblitz/issues/595#issuecomment-530329517

jodobear, 2019-09-23:
After reporting successful on-chain recovery, which did not contain any off-chain recovered funds (see [#595 (comment)](https://github.com/rootzoll/raspiblitz/issues/595#issuecomment-533894645)) I proceeded with recovering off-chain funds following the [recovery.md](https://github.com/lightningnetwork/lnd/blob/master/docs/recovery.md#recovering-using-scbs) as suggested above using following command:

`lncli restorechanbackup --multi_file ~/files/channel.backup`

I got the following output:

`[lncli] unable to restore chan backups: rpc error: code = Unknown desc = unable to unpack chan backup: channel already exists`

I do not see any channels, nor funds on my on-chain wallet. `lncli getinfo` currently shows:

```
...
"num_pending_channels": 0,
"num_active_channels": 0,
"num_inactive_channels": 0,
...
```

rootzoll, 2020-03-28:
Marked for Hackathon Sprint:

- Find solution for better on-chain fund recovery

rootzoll, 2020-04-16:
TODO: After the old wallet was created from SEED or SEED & SCB ... make sure to have it synced to chain before doing a restart. Otherwise it will work after restart with recovery-window of 0 and not find any funds.

TODO: try to give SCB file as byte-array in python call.

closed 2020-04-18 by rootzoll