raspiblitz/raspiblitz #788: Raspiblitz 1.3: Mount HDD: system call failed: Structure needs cleaning.

Geranni, 2019-09-18:
Hi,

After flashing my SD card with 1.3 and trying to start up, I get the following error:

[image]

When I use the command "sudo shutdown -r now" to restart the Raspiblitz I only get to this screen and cannot access via ssh:

[image]

Is there any way to get the installation running? I flashed the image on the SD card several times and it didn't change anything.

fluidvoice, 2019-09-19:
you need to run this command (replace /dev/sda1 with whatever is your HDD device):
`sudo fsck.ext4 -y /dev/sda1`
But you need to SSH or get to a shell prompt. You can do this connecting the HDD to another computer or login here with a connected keyboard and entering a password or `Control-D`

Geranni, 2019-09-23:
Thank you, that brought me a little bit further. Now there is another error message popping up:

[image]

Geranni, 2019-09-23:
Respectively:

[H[2J[3J
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.3
chainnetwork: bitcoin / main
 12:43:50 up  2:52,  2 users,  load average: 0.18, 0.18, 0.12

*** BLOCKCHAIN SYSTEMD STATUS ***

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2019-02-14 10:12:00 GMT, end at Mon 2019-09-23 12:43:50 BST. --
-- No entries --
1569238967 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log

*** LND SYSTEMD STATUS ***

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Thu 2019-02-14 10:12:00 GMT, end at Mon 2019-09-23 12:43:50 BST. --
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log

*** LAST 20 RTL LOGS ***
-- Logs begin at Thu 2019-02-14 10:12:00 GMT, end at Mon 2019-09-23 12:43:50 BST. --
-- No entries --

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
raspberryPi=3
powerFAIL=0
powerWARN=0
powerMIN='1237500 microVolt'
tempFAIL=0
tempWARN=2
tempMAX='6550 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.10.34'
tempCelsius='56.9'
uptime=10324
upsStatus='OFF'
startcountBlockchain=0
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

Geranni, 2019-12-18:
I formatted the HDD a few months ago and my node (Raspberry Pi 3) spent the last 2 months syncing the BTC blockchain from the network directly to avoid potential errors from torrents. Now after starting up I encountered exactly the same problems again... So it seems it does not even work to set up the system completely new. I am sorry, but I do not think that the project Raspiblitz is working. I have been spending probably a week working time in total trying to set up the node and failed. I wish you all the best with your project, but I am out now...

openoms, 2019-12-18:
This is very likely to come down to a common power supply issue, when you disk has not enough power supplied through the USB2 port of the RPi3 which results into filesystem corruption.

Whichever project you are trying to use next I recommend to format and test your disk connected to a PC to make sure it is not damaged physically and use an external source of power through a Y-cable, powered USB hub or external disk case with the RPi3.

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by Geranni