raspiblitz/raspiblitz #790: new update to 1.3 wants to format HDD

Lupus-Pluvia, 2019-09-20:
hello have the following problem the new update wants me to format the hard drive and set up the node again.  how can i handle this or who i am doing this how can i counter this without backup
[image]

rootzoll, 2019-10-02:
This is unusual. There is a check if the HDD was mounted `df | grep -c /mnt/hdd` and if not it thinks it offers that dialog. But maybe we have to even jump a step before that ... normally on a update it should automatically go into recover-mode. So what was the message on the LCD before you SSHed into the RaspiBlitz?

Lupus-Pluvia, 2019-10-02:
According to PC expert, the HDD should be broken and can not read or format

closed 2019-11-16 by rootzoll