raspiblitz/raspiblitz #794: Clarification: When building SD card should HDD be connected?

frennkie, 2019-09-22:
I was building a (slightly) customized SD card images today. I saw at least one message that indicated that `/mnt/hdd` was expected to exist.

https://github.com/rootzoll/raspiblitz/blob/master/README.md#build-the-sd-card-image is not exactly clear on this.

frennkie, 2019-09-22:
Example:

```
*** Prepare TOR source+keys ***
/home/admin/config.scripts/internet.tor.sh: line 17: /home/admin/raspiblitz.info: No such file or directory
/home/admin/config.scripts/internet.tor.sh: line 18: /mnt/hdd/raspiblitz.conf: No such file or directory
```

rootzoll, 2019-10-02:
The HDD should not be connected on sdcard build. 

To avoid that error ... the code block in the `internet.tor.sh` script:
```
source /home/admin/raspiblitz.info
source /mnt/hdd/raspiblitz.conf
if [ ${#network} -eq 0 ]; then
 echo "FAIL - missing /mnt/hdd/raspiblitz.conf"
 exit 1
fi
```

should be moved to line 177 (after if check for prepare). 

TODO once v1.4 branch is open.

frennkie, 2019-12-28:
@rootzoll isn't this fixed..?!

rootzoll, 2020-01-06:
@frennkie yepp this was fixed. CLosing issue.

closed 2020-01-06 by rootzoll