raspiblitz/raspiblitz #796: FanShim, LCD screen Blinking won't finish boot

Aki113s, 2019-09-24:
I posted on FanShim forums, this is to see if anyone else has this issue with a Pi 4, Raspiblitz 1.3

I turn it on and the LCD comes up to boot then half way in booting the screen flickers showing random parts of the screen with the boot sequence showing. No SSH possible.

Took screen off, booted was able to SSH in. I was using a USB fan while awaiting a FanShim, I may go back to that sadly. SO nice to have a screen

Thanks for any insights.

Tracachang, 2019-09-24:
Ever had some screen problems as you mention but I was able to ssh.

What worked for me was flashing the SD card, installed raspbian lite and run the script (manually build ad card) and then screen was working normal.

gandreottola, 2019-09-28:
I had the same problem using fan shim with RP4, Software boots up perfectly fine when you don't have the fan shim. Switched the fan shim with a usb fan because I want both the LCD and fan on my raspberry pi 4, now it works fine.

ZeusMinos, 2019-10-01:
I can confirm that the RP4 board is getting stock at booting and the LCD wont work properly if the Shim Fan is in place. It the Shim service try to start.

I will contact the Shim fan guys to try to debug this issue

gandreottola, 2019-10-01:
When do you think this issue would get resolved?

openoms, 2019-10-03:
To me this seems to be an electrical contact issue, so trying to physically reposition the shim under the LCD might help.
In any case please let know the actual designers of the fan too on this forum:
https://forums.pimoroni.com/t/3-5-lcd-wont-work-with-fan-shim/11687/8

gandreottola, 2019-10-03:
I recently ordered a new type LCD that is has an installed fan, this only made for the raspberry pi 4. Just tested it out with raspiblitz and I am able to boot up the software with no issues. Who ever has a raspberry pi 4, and want to have both fan and LCD on the pi then I recommend getting an LCD with built-in fan, this is just an option for the ones with RP4 and are having problems with fan shim, as well makes the whole raspberry pi set up look very clean.

Link for LCD: https://www.amazon.com/Raspberry-Cooling-Heatsink-320x480-Monitor/dp/B07WRV48ZW/ref=sr_1_2?keywords=lcd+built-in+fan&qid=1570126606&s=electronics&sr=1-2

openoms, 2019-10-03:
@gandreottola nice suggestion, thanks. I like this all round option, having a fan power connector on the back of the screen is a great idea also a spacer to keep the screen stable is a nice touch I was looking into as well. 
Can you confirm that this screen works out of the box with the RaspiBlitz image or you need to install the suggested driver manually?

gandreottola, 2019-10-03:
Yes, I've tested it with the pre-installed raspiblitz image, no need to do it manually.

KhimairaCrypto, 2019-10-05:
> 
> 
> To me this seems to be an electrical contact issue, so trying to physically reposition the shim under the LCD might help.
> In any case please let know the actual designers of the fan too on this forum:
> https://forums.pimoroni.com/t/3-5-lcd-wont-work-with-fan-shim/11687/8
You are !00% right. The connections on the fan are the main issue. The fan cant handle a HAT. Sometimes if the connectors are not well aligned the Pi wont boot. I am remove the LCD and using keeping only the fan. I rather keep the PI cool

fluidvoice, 2019-10-09:
> I recently ordered a new type LCD that is has an installed fan, this only made for the raspberry pi 4. Just tested it out with raspiblitz and I am able to boot up the software with no issues. Who [...]

yes looks clean. What kinda temps do you see during and after syncing?

[image]

KhimairaCrypto, 2019-10-09:
The issue with that type of case is that the air flow is not that great, because of the full enclosure!

fluidvoice, 2019-10-09:
This or similar LCD with fan and power-pins on the rear is for sale without the case, so *if* it's the same size as other 3.5" screens it might allow other cases, for example the CryptoCloaks LightningShell:

[image]

https://www.amazon.com/Raspberry-Resistive-Heat-Sinks-320x480-RASPBIAN/dp/B07VHQX37S

openoms, 2019-10-09:
Nice find! Ordered to test: https://www.amazon.co.uk/Raspberry-Resistive-heat-sinks-320x480-RASPBIAN/dp/B07VHQWG37

fluidvoice, 2019-10-09:
> The issue with that type of case is that the air flow is not that great, because of the full enclosure!

[image]

fluidvoice, 2019-10-09:
> Nice find! Ordered to test: https://www.amazon.co.uk/Raspberry-Resistive-heat-sinks-320x480-RASPBIAN/dp/B07VHQWG37

Just pinged CryptoCloaks, he's buying to test too.

gandreottola, 2019-10-09:
> yes looks clean. What kinda temps do you see during and after syncing?

Yea, the case makes the pi run hot by keeping all the heat inside but I took out the case and just left the bottom half of the case, right now my pi is running at 47 celsius.

[image]

openoms, 2019-10-09:
@gandreottola, good number! I think in any kind of active cooling the RPi4 is comfortable. Also it is veri cheap to reaplace thia kind of fan requiring on two pins (especially vs the shim fan.)

I see you have the touch pen on the side. 
Did you test the RaspiBlitz touch function yet?
The experimental option is to be activated in the SERVICE menu and can be deactivated any time. Requires a restart. 
Calibration help here: https://github.com/rootzoll/raspiblitz/issues/809#issuecomment-538871135

gandreottola, 2019-10-09:
I have not tested the RaspiBlitz touch function, but thinking to try it out later on.

KhimairaCrypto, 2019-10-10:
My findings so far:
https://forums.pimoroni.com/t/3-5-lcd-wont-work-with-fan-shim/11687/13?u=khimaira

openoms, 2019-10-10:
> Nice find! Ordered to test: https://www.amazon.co.uk/Raspberry-Resistive-heat-sinks-320x480-RASPBIAN/dp/B07VHQWG37

@fluidvoice The new screen is working straight out of the box. Doesn't even need new drivers, just the one already installed. My broken touch function is alive too (probably my old LCD was broken). The fan is spinning, temp at 53 C. The fan is glued to the GPIO connector, but probably could come of if enough force or a knife applied.

fluidvoice, 2019-10-11:
Thanks. Will order one.

fluidvoice, 2019-10-11:
> @fluidvoice The new screen is working straight out of the box. Doesn't even need new drivers, just the one already installed. My broken touch function is alive too (probably my old LCD was broken). [...]

Thanks. Will order one. Will this be added onto any of the shopping lists?

openoms, 2019-10-20:
if anyone has a problem with the shim fan, just manually bend the GPIO out  very slightly (like 1mm) to make all the connections tight.
My new screen and shim fan pair stopped flickering instantly after I have done this. It is confirmed to be a contact issue.

Aki113s, 2019-10-29:
> 
> 
> if anyone has a problem with the shim fan, just manually bend the GPIO out very slightly (like 1mm) to make all the connections tight.
> My new screen and shim fan pair stopped flickering instantly after I have done this. It is confirmed to be a contact issue.

I would try but my fanshim died :( I for an enclosed case with fan built on LCD, as expected it runs hotter than I like but not throttling. LCD works.

hermanlim, 2019-11-10:
> I posted on FanShim forums, this is to see if anyone else has this issue with a Pi 4, Raspiblitz 1.3
>
> I turn it on and the LCD comes up to boot then half way in booting the screen flickers showing [...]

Plymouth is causing problems. I had the exact same problem as you. The Raspberry will boot eventually but this could take 20 minutes to 6 hours. The easiest way to tell is to watch the ethernet port lights (you will be able to ssh in when it is blinking). Once you are in, uninstall plymouth (  sudo apt-get purge plymouth*  ) and all will be fine.

--Herman

KhimairaCrypto, 2019-11-13:
@hermanlim you are my hero :-O. I tried almost everything. How did you findout that this was the issue?
It works  🥇

hermanlim, 2019-11-13:
I took a 'live' photo of the boot screen as it was flashing through to see
what the error message was and noticed there was an error pertaining to
"Plymouth'. After that, looked through the log files and then searched some
online forums and found plenty of plymouth mischief!

openoms, 2019-11-13:
Great find @hermanlim! 
I think we should weed out plymouth during the SDcard build process. It would be only necessary for the desktop GUI which we don't use.

f-daniel, 2019-11-14:
just encountered the same problem and after trying just about all the suggestions in the thread simply removing plymouth worked, thanks @hermanlim!

rootzoll, 2019-11-15:
Open up an issue for 1.4 release: Fix BootUp Problems - Remove plymouth #861
Thx @hermanlim for the finding.

@openoms & @fluidvoice how was the testing of the LCD+Fan? Can we plan to update shopping lists for v1.4 release to that to reduce ShimFan connection problems?

openoms, 2019-11-15:
> @openoms & @fluidvoice how was the testing of the LCD+Fan? Can we plan to update shopping lists for v1.4 release to that to reduce ShimFan connection problems?

I am still pro ShimFan (especially that now the flickering problem seem to have a software fix) manly because it does not run all the time. I have found it necessary during the initial Block Download, but during the day-to-day operation I prefer having the fan off for up to 70 degrees.
set up like this (https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-fan-shim)
```
git clone https://github.com/pimoroni/fanshim-python
cd fanshim-python
sudo ./install.sh

cd examples
sudo ./install-service.sh --on-threshold 70 --off-threshold 55 --delay 2
```
With the LCD-with-powerpins-and-fan I found:
pro:
+ it works
+ cheaper than LCD + shimfan
+ fan is replaceable (other fan can be attached to the pins)

cons:
- noisy and always on - can be only controlled physically by unplugging.
- needs to be broken off to replace

I am looking forward to test some big heatsinks with the RPI4. With some proper airflow it could be still an option to have a no-moving-part setup.
There are plenty of these appeared since the RPi4 came out: https://www.aliexpress.com/wholesale?SearchText=raspberry+4+heatsink+case . I am waiting to have one posted. and will do some tests.

hermanlim, 2019-11-16:
I’ve ordered it in. I will post an update once I’ve had a chance to put it
through its paces.

openoms, 2019-12-02:
wondering if anyone having solved their shim fan issue has had any further problems? Things like the `sudo reboot` or `sudo shutdown now` not working or else?

It might just affecting the latest build: https://github.com/rootzoll/raspiblitz/issues/885
Also did any luck with chaning the shimfan to a never one? I wonder if this is issue is just a manufacturing error or a faulty update to their PCB?

Removing plymouth might have got through the booting problem but unfortunately not catching the problem early can lead to weird issues later on.

I have tested with a GPIO Expansion board. Physically separating the connection of the shim fan from the LCD did not help my faulty one. Also unlikely to be an EM interference as distance form the board did not make a difference either.

openoms, 2019-12-02:
Also need to test if having the shim-fan software preinstalled would prevent the problems? https://github.com/rootzoll/raspiblitz/issues/806

openoms, 2019-12-04:
My faulty shim-fan stops a new RPi4 too from shutting down or rebooting.

Installing shim-fan software does not make a difference.

Will order another shim fan to see if this is a consistent issue.

hermanlim, 2019-12-05:
the not shutting down is due to the bitcoind not quitting. The system will
shut down if you issue the command with the 'force' option though it will
have issues booting later due to the need for the blockchain to sync
properly.

openoms, 2019-12-05:
> the not shutting down is due to the bitcoind not quitting. The system will shut down if you issue the command with the 'force' option though it will have issues booting later due to the need for the blockchain to sync properly.

Possible, but when I remove the faulty shim-fan the `sudo reboot` or `sudo shutdown now` works instantly. How would the shim-fan affect the bitcoind? Will test with `force` and update here. Still would be very strange to need that.

This is the weird behaviour with long time in between the commands.
```
admin@RaspiBlitz:~ $ sudo shutdown now
admin@RaspiBlitz:~ $ sudo reboot
admin@RaspiBlitz:~ $ 
```

more described in here: https://github.com/rootzoll/raspiblitz/issues/885

hermanlim, 2019-12-05:
Hmm. I must admit, I did not try it with the shim-fan physically removed as I did not consider the shim-fan interfering with the shutting down. 

Can you try quitting bitcoind manually first and then try shutting down? See if it makes any difference? Perhaps that’s not the culprit in your case and we are barking up the wrong tree!

What worked all the time was the reboot -f but this lead to problems on the actual reboot (basically got stuck on the blue screen on reboot and not being able to proceed to the menu) and I had to reindex the chain manually to fix it.

openoms, 2019-12-05:
I can confirm that it is a shim-fan causing the problem and using `sudo reboot -f` is certainly not a viable solution.

! got a new shim-fan today and unfortunately it produces the same problem so 2 out of 3 of mine are faulty (only the one from an early batch is ok) although there is no visual difference between them.

I tested 3 screens and 2 RPi4-s, and the above mentioned 3 fans in various combinations.

With the faulty fans the problem with plymouth and the shutdown issues are reproducable. 
When I remove the fan and reboot (need to pull the plug) everything works again (including instant `sudo reboot`, `sudo shutdown now` and no screen flickering when plymouth is reinstalled)
When I put a faulty fan back and all the issues appear again.

Will send this new fan of mine back to pimoroni (bought through Amazon).

thelightningist, 2019-12-20:
Purging Plymouth worked like a charm. Now the shim fan and the LCD run in my cryptocloak without issue

rootzoll, 2020-01-28:
OK v.14 will contain this fix. Closing issue for now.

closed 2020-01-28 by rootzoll