raspiblitz/raspiblitz #800: LND Wallet RED. Exhausted efforts to green or even yellow up.

IW8, 2019-09-26:
[image]

Port Forwarding did not work. Everything was red.

[image]

Probably because I am in a hotel using my router on their spectrum internet going back to the CO to a Nomadix switch that does not allow port forwarding rules.

UPNP does not work either. Everything was red.

TP-Link router did offer Dynamic DNS. mh5.tplinkdns.com which did yellow up Bitcoin node.

Looking for assistance in greening up LND.

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.3
chainnetwork: bitcoin / main
 00:06:29 up 1 day, 20:48,  2 users,  load average: 1.42, 1.32, 1.00

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-09-25 03:20:57 BST; 1 day 20h ago
  Process: 3657 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 3663 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 3714 (bitcoind)
    Tasks: 18 (limit: 2200)
   Memory: 481.2M
   CGroup: /system.slice/bitcoind.service
           └─3714 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitc…

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2019-09-26 19:00:42 BST, end at Fri 2019-09-27 00:06:29 BST. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-09-26T20:33:09Z UpdateTip: new best=00000000000000000015d2e26e7496ee06e8c3b783ece165f1d60c38923b23b8 height=596736 version=0x20000000 log2_work=91.137689 tx=459191146 date='2019-09-26T20:33:21Z' progress=1.000000 cache=62.3MiB(550227txo) warning='48 of last 100 blocks have unexpected version'
2019-09-26T20:44:12Z Pre-allocating up to position 0x300000 in rev01808.dat
2019-09-26T20:44:13Z UpdateTip: new best=000000000000000000081cc7306957d1abdf1597274c1e116c8adf5a4784323a height=596737 version=0x3fff0000 log2_work=91.137718 tx=459194184 date='2019-09-26T20:43:49Z' progress=1.000000 cache=62.7MiB(553842txo) warning='49 of last 100 blocks have unexpected version'
2019-09-26T20:50:48Z Pre-allocating up to position 0x2000000 in blk01808.dat
2019-09-26T20:50:49Z UpdateTip: new best=0000000000000000000b5d4144eeb37300519c253ac7a6d3513631f23f746255 height=596738 version=0x20000000 log2_work=91.137747 tx=459197094 date='2019-09-26T20:50:45Z' progress=1.000000 cache=63.0MiB(556696txo) warning='48 of last 100 blocks have unexpected version'
2019-09-26T20:57:49Z UpdateTip: new best=00000000000000000011d3ee99f34932f81459cdb1e2c50359a8a9ce0476acba height=596739 version=0x3fff0000 log2_work=91.137776 tx=459200245 date='2019-09-26T20:58:15Z' progress=1.000000 cache=63.4MiB(559975txo) warning='48 of last 100 blocks have unexpected version'
2019-09-26T21:00:42Z UpdateTip: new best=0000000000000000000eed1889757f158cde36115a27609eb7c87971ad906f85 height=596740 version=0x3fff0000 log2_work=91.137805 tx=459201696 date='2019-09-26T21:00:49Z' progress=1.000000 cache=63.5MiB(561566txo) warning='48 of last 100 blocks have unexpected version'
2019-09-26T21:06:41Z UpdateTip: new best=0000000000000000000445d7bc8c2a4944793dcb7fa37c0a023aa0f223258dc0 height=596741 version=0x20c00000 log2_work=91.137834 tx=459203246 date='2019-09-26T21:06:24Z' progress=1.000000 cache=63.7MiB(563517txo) warning='48 of last 100 blocks have unexpected version'
2019-09-26T21:21:16Z UpdateTip: new best=000000000000000000119b42cb6faed8c44986d2229ad36f241cfea05044105e height=596742 version=0x20002000 log2_work=91.137863 tx=459206228 date='2019-09-26T21:20:51Z' progress=1.000000 cache=64.3MiB(568548txo) warning='48 of last 100 blocks have unexpected version'
2019-09-26T21:22:23Z UpdateTip: new best=00000000000000000002bb4c1e5cd45ac7ad013dc556176f6ca1667b9ff41396 height=596743 version=0x20400000 log2_work=91.137892 tx=459207430 date='2019-09-26T21:22:15Z' progress=1.000000 cache=64.3MiB(568987txo) warning='48 of last 100 blocks have unexpected version'
2019-09-26T21:33:33Z UpdateTip: new best=0000000000000000000e1348d913adf4c452a4a063eaeba338a33f4aa20448e3 height=596744 version=0x20000000 log2_work=91.137921 tx=459210223 date='2019-09-26T21:33:07Z' progress=1.000000 cache=65.0MiB(575201txo) warning='48 of last 100 blocks have unexpected version'
2019-09-26T21:34:09Z Pre-allocating up to position 0x400000 in rev01808.dat
2019-09-26T21:34:09Z UpdateTip: new best=00000000000000000003701784373483efe39f8afb91b862061d3dc1f4a1dd46 height=596745 version=0x20400000 log2_work=91.13795 tx=459210586 date='2019-09-26T21:33:37Z' progress=1.000000 cache=64.9MiB(574538txo) warning='49 of last 100 blocks have unexpected version'
2019-09-26T21:34:48Z UpdateTip: new best=000000000000000000085e4ea644892ccf2b4082866e48d4b718cf83ab2d20a3 height=596746 version=0x20000000 log2_work=91.137979 tx=459210850 date='2019-09-26T21:34:40Z' progress=1.000000 cache=65.0MiB(574858txo) warning='49 of last 100 blocks have unexpected version'
2019-09-26T21:50:39Z UpdateTip: new best=0000000000000000000bc79d523896c2e52355d65390df402a67f297ef2f4ff3 height=596747 version=0x20000000 log2_work=91.138008 tx=459214361 date='2019-09-26T21:50:31Z' progress=1.000000 cache=65.5MiB(579845txo) warning='49 of last 100 blocks have unexpected version'
2019-09-26T21:56:35Z UpdateTip: new best=00000000000000000015bdf9982127838b0ec0200773099b17aa4ceb72f411d2 height=596748 version=0x20000000 log2_work=91.138037 tx=459216134 date='2019-09-26T21:55:12Z' progress=0.999999 cache=65.6MiB(581209txo) warning='49 of last 100 blocks have unexpected version'
2019-09-26T22:12:07Z UpdateTip: new best=000000000000000000153236db55dc539fd9ac08a6879ee52ae36d904078b0a2 height=596749 version=0x3fff0000 log2_work=91.138066 tx=459218415 date='2019-09-26T22:11:58Z' progress=1.000000 cache=66.0MiB(584995txo) warning='50 of last 100 blocks have unexpected version'
2019-09-26T22:30:25Z UpdateTip: new best=00000000000000000001546543588aba7ef3c0b1f01f342ca5d14c016991fb18 height=596750 version=0x20000000 log2_work=91.138095 tx=459221797 date='2019-09-26T22:29:55Z' progress=1.000000 cache=66.6MiB(590045txo) warning='49 of last 100 blocks have unexpected version'
2019-09-26T22:32:54Z UpdateTip: new best=000000000000000000145e42fce1b9b0247547e8dadedeae83fb7c5005c854fc height=596751 version=0x20000000 log2_work=91.138124 tx=459223903 date='2019-09-26T22:32:54Z' progress=1.000000 cache=66.7MiB(591088txo) warning='49 of last 100 blocks have unexpected version'
2019-09-26T22:34:25Z UpdateTip: new best=0000000000000000000c82e4a2f3f881d1f7f7d0f4bcb78a4d4ddafd87987984 height=596752 version=0x7fffe000 log2_work=91.138153 tx=459224186 date='2019-09-26T22:33:45Z' progress=1.000000 cache=66.7MiB(591507txo) warning='49 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-09-25 03:25:53 BST; 1 day 20h ago
  Process: 18107 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 18109 (lnd)
    Tasks: 17 (limit: 2200)
[... 60 lines cut ...]
tempCelsius='36.4'
uptime=161301
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=596752
initialSync=0
syncProgress=100.00
startcountLightning=5
lndActive=1
walletLocked=0
syncedToChain=1
scanTimestamp=1569537225
scanDate='Thu 26 Sep 23:33:45 BST 2019'
scanProgress=99.99
lndRPCReady=1
scriptRuntime=2

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

Press ENTER to return to main menu.
Thank You.

openoms, 2019-10-03:
You can use Tor to get around the firewalls you have no access to open.

IW8, 2019-10-06:
Activated Tor option in Services. It greened everything up.
[image]
Thank You.

closed 2019-11-16 by rootzoll