raspiblitz/raspiblitz #802: Tor fail after system update

pg024, 2019-09-28:
Hello, I have just performed a raspbian system update and TOR just failed and won't start anymore. I am on the latest raspiblitz release.
output from `$journalctl -u tor@default`

`Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.002 [notice] Tor 0.4.1.6 running on Linux with Libevent 2.1.8-stable, OpenSSL 1.1.1c, Zlib 1.2.11, Liblzma 5.2.4, and Libzstd 1.3.8.
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.002 [notice] Tor can't help you if you use it wrong! Learn how to be safe at https://www.torproject.org/download/download#warning
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.002 [notice] Read configuration file "/usr/share/tor/tor-service-defaults-torrc".
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.002 [notice] Read configuration file "/etc/tor/torrc".
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.016 [warn] Skipping obsolete configuration option 'PortForwarding'
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.020 [notice] Opening Socks listener on 127.0.0.1:9050
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.020 [notice] Opened Socks listener on 127.0.0.1:9050
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.020 [notice] Opening Control listener on 127.0.0.1:9051
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.020 [notice] Opened Control listener on 127.0.0.1:9051
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.000 [warn] Couldn't open file for 'Log notice file /mnt/hdd/tor/notice.log': Read-only file system
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.000 [warn] Couldn't open file for 'Log info file /mnt/hdd/tor/info.log': Read-only file system
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.000 [notice] Closing partially-constructed Socks listener on 127.0.0.1:9050
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.000 [notice] Closing partially-constructed Control listener on 127.0.0.1:9051
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.000 [warn] Failed to parse/validate config: Failed to init Log options. See logs for details.
Sep 28 23:31:43 redpenguin tor[674]: Sep 28 23:31:43.000 [err] Reading config failed--see warnings above.
Sep 28 23:31:43 redpenguin systemd[1]: tor@default.service: Main process exited, code=exited, status=1/FAILURE
Sep 28 23:31:43 redpenguin systemd[1]: tor@default.service: Failed with result 'exit-code'.
Sep 28 23:31:43 redpenguin systemd[1]: Failed to start Anonymizing overlay network for TCP.
Sep 28 23:31:43 redpenguin systemd[1]: tor@default.service: Service RestartSec=100ms expired, scheduling restart.
Sep 28 23:31:43 redpenguin systemd[1]: tor@default.service: Scheduled restart job, restart counter is at 1.
Sep 28 23:31:43 redpenguin systemd[1]: Stopped Anonymizing overlay network for TCP.
Sep 28 23:31:43 redpenguin systemd[1]: Starting Anonymizing overlay network for TCP...
`

Could be it a data disk failure? This is the 2nd time happening on a fresh raspiblitz install after weeks of running smoothly.
Thank you in advance.

openoms, 2019-09-29:
What I can make out of this that `'Log info file /mnt/hdd/tor/info.log': Read-only file system` would mean a disk problem indeed.
There are 2 common causes:
* Power supply issue - should be sorted with separate power supply to the HDD like a powered USB Hub or Y-cable with a separate power supply. Upgrade to a better >3A PSU if not using one already.
* Physical disk error - you will need a new HDD and start from scratch.

In any case if you can login via ssh run `/home/admin/99checkUpdate.sh` and select `Update anyway` to have the script guide you through an LND backup process.

pg024, 2019-09-29:
Thank you for the answer.
I am not sure about the physical disk error, it's a quite new nano SSD (6 months old, Samsung). Can't I just delete info.log and let the tor service recreate it?
ps I can run normally the node without TOR. 

Brad on the TG chat last time said the issue could be related to this, but I did not dig into:

https://imgur.com/GTtVKbv.png

openoms, 2019-09-30:
Rpi4 or 3? Is it the Samsung T5 disk?
You could deactivate Tor from the menu and delete the whole Tor directory from the SSD.
When you reactivate it the directory will be recreated.

pg024, 2019-09-30:
Rpi3, nope it is a Samsung SSD 860 msata/nano disk.
I have tried to delete the whole tor directory, unfortunately the problem persists.

These are the tor related packages that have been upgraded on the last system update.
`tor-geoipdb:armhf (0.4.1.5-1~d10.buster+1, 0.4.1.6-1~d10.buster+1)`
`tor:armhf (0.4.1.5-1~d10.buster+1, 0.4.1.6-1~d10.buster+1)`

I can try to revert them to previous versions, though.

pg024, 2019-09-30:
`Sep 30 12:15:06 redpenguin tor[682]: Sep 30 12:15:06.496 [notice] Opening Control listener on 127.0.0.1:9051`
`Sep 30 12:15:06 redpenguin tor[682]: Sep 30 12:15:06.496 [notice] Opened Control listener on 127.0.0.1:9051`
`Sep 30 12:15:06 redpenguin tor[682]: Sep 30 12:15:06.498 [warn] Error creating directory /mnt/hdd/tor/sys: No such file or directory`
`Sep 30 12:15:06 redpenguin tor[682]: Sep 30 12:15:06.498 [warn] Failed to parse/validate config: Couldn't create private data directory "/mnt/hdd/tor/sys"`
`Sep 30 12:15:06 redpenguin tor[682]: Sep 30 12:15:06.498 [err] Reading config failed--see warnings above.`
`Sep 30 12:15:06 redpenguin systemd[1]: tor@default.service: Main process exited, code=exited, status=1/FAILURE`
`Sep 30 12:15:06 redpenguin systemd[1]: tor@default.service: Failed with result 'exit-code'.
`

Disabling/renabling tor service does not create the tor folder on /mnt/hdd/ anymore. Any hint on this? I am gonna try unistalling tor packages and let the script handle it.

pg024, 2019-09-30:
I believe I break my system-tor lol. 
`$sudo journalctl -u tor@default`
`-- Logs begin at Mon 2019-09-30 12:23:29 BST, end at Mon 2019-09-30 12:30:37 BST`
`-- No entries --`

Do you suggest to re-flash a new raspiblitz install?

openoms, 2019-09-30:
Sure that is the fallback when the settings are messed up.
Backup LND (if there are funds) with the method above and start with a clean 1.3 SDcard image which we know  is working.

closed 2019-10-01 by pg024