raspiblitz/raspiblitz #803: Rainbow screen on first boot

operahouserat, 2019-09-29:
When trying to boot my Raspberry Pi 4 for the first time after writing the SD card image to the SD card, this rainbow screen appears. I've now tried two different SD cards and have the same issue.

Any suggestions?

[image]

operahouserat, 2019-09-29:
Think it was a power supply issue.

closed 2019-09-29 by operahouserat

Avdhoot28, 2021-04-01:
> When trying to boot my Raspberry Pi 4 for the first time after writing the SD card image to the SD card, this rainbow screen appears. I've now tried two different SD cards and have the same issue.
> 
> Any suggestions?
> 
> [image]

> Did you got any solution/ Did your raspberry pi 4 got booted successfully ?